jopohl/urh #36: Change label concept from groupwise to blockwise

andynoack, 2016-05-17:
Why this?
------

Complex protocols have messages with different lengths, groupwise labels do not fulfill this requirement when there are different message types in a block. Separating such messages to different groups would destroy the time connection, therefore this is not a solution.

Additionally blockwise labels make fuzzing more generic.

Todo
--------------

Ergonomic concept to attach labels, e.g. rules (cf. wireshark), has to be worked out.

jopohl, 2016-05-22:
I see two possibilities for achieving this. First, create labelsets __per group__ or second create labelsets __globally__. Unfortunately, both variants have some drawbacks.

## Group based labelsets
Using the exisiting protocol groups to classify the labelsets looks like the best way. However, this complicates the labeling process for the end-user. There are three ways to present this:

[image]

- Option 1 is a three level tree which looks way too overkill.
- Option 3 works with comboboxes. This is the most easy to implement but requires messing with two comboboxes to get to the label of interest.
- Option 2 is a hybrid of 1 and 3.

## Global labelsets
Having global labelsets will make the labeling process much easier. The GUI may look like this:

[image]

Option 2 uses a single combobox and looks quite nice. It will also be, by far, the easiest to implement. However, having global labelsets leads to the question what the protocol groups are for. The logical connection between protocol group and label set gets lost. Nevertheless, since labelsets can be set per block the logical separation can also be achieved with global labelsets.

## Summary
Group based labelsets | Global labelsets
------------ | -------------
:+1:  Integrates with existing protocol groups | :+1:  Easy to understand for the end-user and easier to implement
:-1:  More complex implementation and useability | :-1: Raises question about the relevance of protocol groups

I think the main problem is, that labelsets bring a new logical level to protocol analysis that somehow collides with the existing protocol groups.

andynoack, 2016-05-24:
Decision: We go for global labelsets, as group based labelsets are covered implicitely.

closed 2016-06-01 by jopohl