jopohl/urh #86: Implement CRC plugin

andynoack, 2016-08-09:
* New window
* Choose or define CRC polynomial
* Set CRC position (absolute or relative to the end)
* Choose range for input text (on basis of labels, or starting label + crc position)
* Return True/False

psa-jforestier, 2017-02-22:
Have a look at the CRC RevEng (http://reveng.sourceforge.net/) tool : a portable C program to compute CRC from an arbitrary input stream (can be an hex string on the command line) ; also have the hability to search for a CRC algorithm based on two or more input pattern (usefull to reverse engeneering an unknown protocol). Maybe it can be implemented as a Plugin in URH.

andynoack, 2017-02-22:
Thank you for this advice! Indeed there is already an algorithm in the code that reverse engineers CRCs with the idea of Gregory Ewing (http://www.cosc.canterbury.ac.nz/greg.ewing/essays/CRC-Reverse-Engineering.html), but unfortunately I have not implemented a GUI (will probably be a plugin) for that yet. 

The problem with reverse engineering of CRCs is that it is very time consuming for larger polynomials which stands against the responsiveness of the GUI. This holds also for reveng. However, providing a plugin for CRC is an important point on our todo list :)

PR #280 (2017-07-02): checksum module
Goal of this PR is to fix #86 .

We want to make the CRC function more customizable and show CRC errors with colors in the GUI instead only XORing calculated CRC with present bits. For this, following steps are necessary:

- [x] Program backend functions for CRC checking @andynoack 
- [x] Create a draft for CRC configuration dialog @jopohl
- [x] Find a place from where to open CRC configure dialog @andynoack  and  @jopohl 
   -  In label editing dialog below table: New tab widget. Tab names are label names (if type == crc)
   - if no tab -> hide tab widget
- [x] Sync CRC settin [... cut]
closed 2017-07-02 by jopohl