jopohl/urh #105: Crashes when using analyse button if external program for encoding/decoding is missing

andynoack, 2016-10-27:
ime for generating UI: 1.19 seconds
Using modules from /home/ryu/util/Eigene Dateien/Subversion/urh/src
[DEBUG] Time for auto assigning participants: 0.24841642379760742
/bin/sh: ./decodings/enocean_switchtelegram: Datei oder Verzeichnis nicht gefunden
[...]
[DEBUG] Time for auto assigning decodings: 0.42714571952819824
[DEBUG] Time for auto assigning message types: 0.001861572265625
Traceback (most recent call last):
  File "/home/urh/src/urh/controller/CompareFrameController.py", line 1265, in on_btn_analyze_clicked
    self.proto_analyzer.auto_assign_labels()
  File "/home/urh/src/urh/signalprocessing/ProtocolAnalyzer.py", line 850, in auto_assign_labels
    label_assigner.auto_assign_to_message_type(self.default_message_type)
  File "/home/urh/src/urh/signalprocessing/LabelAssigner.py", line 201, in auto_assign_to_message_type
    sync = self.find_sync()
  File "/home//urh/src/urh/signalprocessing/LabelAssigner.py", line 123, in find_sync
    sync_interval = max(possible_sync_pos, key=possible_sync_pos.__getitem__)
ValueError: max() arg is an empty sequence

Process finished with exit code 134

Signal was enocean*

jopohl, 2016-10-28:
This is still the old analyze feature, that will be replaced by AWRE. I will perform a merge of the AWRE branch soon (28.10. or 29.10.), where this should be fixed.

jopohl, 2016-10-28:
Fixed in current master. Also added a unittest for this, to prevent this from happening again.

closed 2016-10-28 by jopohl