jopohl/urh #107: Usage

vsboost, 2016-12-05:
Hi there,

Really like the look of this program, just wondering if you could possibly post up a youtube video on general usage with an FSK / ASK signal ?

Thanks

jopohl, 2016-12-05:
Hi  @vsboost,

glad to hear, that you like the program. I will create a little video as soon as I got some time. Probably in the next week.

vsboost, 2016-12-05:
Thanks, look forward to seeing it in action.

jopohl, 2016-12-20:
Hi @vsboost ,

I just uploaded a video for an ASK/FSK signal: https://www.youtube.com/watch?v=QqVvEOzKPCs
More videos will follow soon, but this one should get you started. :-)

vsboost, 2016-12-21:
Thanks for this, excellent video.

vsboost, 2016-12-22:
Ill close this one, your videos are excellent, keep up the great work.

closed 2016-12-22 by vsboost