jopohl/urh #116: Windows compilation error - (errno 2 No such file or directory)

imaurus, 2017-01-25:
Compilation on Windows 10 gave me this error:

`save_to_xml
    with open(constants.FIELD_TYPE_SETTINGS, "w") as f:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\HKEY_CURRENT_USER\\Software\\urh\\fieldtypes.xml'`

As workaround I modified "constants.py". I changed FIELD_TYPE_SETTINGS to "fieldtypes.xml" instead of "os.path.realpath...blah blah blah"... and now it works.

I guess because QSettings on Windows, NativeFormat settings are stored in the following registry paths:

HKEY_CURRENT_USER\Software\MySoft\Star Runner
HKEY_CURRENT_USER\Software\MySoft\OrganizationDefaults
HKEY_LOCAL_MACHINE\Software\MySoft\Star Runner
HKEY_LOCAL_MACHINE\Software\MySoft\OrganizationDefaults

jopohl, 2017-01-25:
Thanks for reporting. a18e7adea720d120ac13ac564a41c68cbeadc3a4 fixes this by using INI Format for QSettings.

jopohl, 2017-01-25:
Fixed with  new [Release](https://github.com/jopohl/urh/releases/tag/v1.2.4.2)

closed 2017-01-25 by jopohl