jopohl/urh #131: Spectrum analyzer - OSError: [Errno 9] Bad file descriptor

kolen, 2017-02-16:
Selected rtl-sdr via gnuradio in settings. File -> Spectrum analyzer, pressing "Record" button, signal plot appears for just a moment then program crashes with following error:

```
▶ urh
Will not regenerate UI, because script cant be found. This is okay in release.
[INFO] Restarting with new port
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/urh/dev/gr/SpectrumThread.py", line 31, in run
    rcvd += recv(32768)  # Receive Buffer = 32768 Byte
OSError: [Errno 9] Bad file descriptor
[1]    57807 abort      urh
```

jopohl, 2017-02-16:
Thanks for reporting @kolen .

I was not able to reproduce this one, but I catched the exception in 9feccea7b93427096c6e35a80b083ceffa809a81 to prevent the program from crashing. Does this work for you? If not: Can you record with your device in Gnuradio itself?

kolen, 2017-02-16:
Yes, tried 'osmocom Source' and 'RTL-SDR Source' blocks in grc — they work. Script from urh works too (I set ``/usr/bin/python`` in settings):

```
▶ /usr/bin/python dev/gr/scripts/rtl-sdr_recv.py --port 11111 --samplerate 2800000
Mac OS; Clang version 8.0.0 (clang-800.0.42.1); Boost_106300; UHD_003.010.000.000-0-unknown

gr-osmosdr ae686c46 (0.1.5git) gnuradio 3.7.10.1
built-in source types: file fcd rtl rtl_tcp uhd rfspace redpitaya
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
Exact sample rate is: 2800000.037087 Hz
[R82XX] PLL not locked!
```

```
▶ nc localhost 11111 |pv > /dev/null
 125MiB 0:00:06 [21.4MiB/s]
```

And now for some reason connect loop stopped working:

```python
        while not self.isInterruptionRequested():
            try:
                self.socket.connect((self.ip, self.port))
                break
            except (ConnectionRefusedError, ConnectionResetError):
                continue
```

For some reason, when retrying failed connect on the same socket, ``OSError: [Errno 22] Invalid argument`` exception is raised.

After adding ``sleep(5)`` before connecting and setting port to another (for some reason 1337 results in 'Address already in use' despite `lsof` telling nothing is using it), it started to work, however it stops receiving after some time.

jopohl, 2017-02-16:
Based on your comments and the discussion on Stackeroverflow you linked, I modified the code to:

```
        while not self.isInterruptionRequested():
            try:
                time.sleep(0.1)
                self.socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
                self.socket.setsockopt(socket.IPPROTO_TCP, socket.TCP_NODELAY, 1)
                self.socket.connect((self.ip, self.port))
                break
            except (ConnectionRefusedError, ConnectionResetError):
                continue
```

Does this also work for you?

kolen, 2017-02-16:
Yes, now it works, both in Spectrum and Record. Thanks.

jopohl, 2017-02-16:
You're welcome!

closed 2017-02-16 by jopohl