jopohl/urh #133: Does urh support USB based USRP such as USRP B200/B210?

an0nym0usatch1na, 2017-02-18:
Loved urh very much, it works fine with my hackrf, but does urh support USB based USRP such as USRP B200/B210? I can only find USRP support of old ip address based version USRP. Thanks.

jopohl, 2017-02-18:
Thanks @an0nym0usatch1na . URH currently offers no support USRP B200/210. I will add support for this soon.

an0nym0usatch1na, 2017-02-18:
nice!!!

jopohl, 2017-02-22:
Hi @an0nym0usatch1na 

I do not possess a USRP B200/210, but it seems to use the same UHD block in Gnuradio. I just fixed the IP Validator to allow address: "". Could you enter "" in IP field send/receive dialog, and see if it works?

dkozel, 2017-02-24:
Hi @jopohl

All USRPs can be found and accessed through the common multi_usrp API. Currently URH is asking for an IP address and hardcoding the "addr=<supplied IP>".
https://github.com/jopohl/urh/blob/master/src/urh/dev/gr/scripts/usrp_recv.py#L33

It would be more generic, and useful, to ask for device arguments. This is the standard approach for the UHD examples and is also used in the GNU Radio (Companion) interface. 
https://github.com/EttusResearch/uhd/blob/master/host/examples/rx_samples_to_file.cpp#L220

UHD documentation of supported keys:
http://files.ettus.com/manual/page_identification.html

There are also various configuration values that can be passed into the same argument string.
http://files.ettus.com/manual/page_configuration.html

I don't have hardware on hand, but will give it a try shortly with a few USRPs. Thank you for all your time producing this tool, I look forward to running it!

jopohl, 2017-02-24:
Hi @dkozel 

thanks a lot for these useful links! That will definitely be helpful for enhancing the send/receive dialogs described in https://github.com/jopohl/urh/issues/148.

cn0xroot, 2017-03-02:
Really need Add support for USRP B200

pwicks86, 2017-03-02:
@jopohl I've tried the newest master (38887125b364cfbe73eb55ecfa95eb2a4ab3206a) and it's still not working with my B200mini using "" as the ip address. Looking at the code, looks like dkozel has it about right.

Also, I've got access to most of the Ettus SDRs and a LimeSDR as well, if you need someone to test those. Finally, as you mentioned, it probably makes sense to develop a UHD interface either with cython or as a c extension, I was thinking along the same lines myself, if you are interested in accepting PR's or need a hand.

jopohl, 2017-03-02:
@pwicks86 that would be awesome! My idea with entering "" as IP was only a guess, because I lack USRP B200 for testing, so it would be very cool if you could support us with testing it.

I am glad for any help writing a Cython Wrapper for UHD, because my time is quite limited right now and would of course accept a PR. Let me know if you need any further information on this!

dkozel, 2017-03-03:
Here's a minimal changeset which allows the device arguments to be passed directly through for reception. I've tested with B200, B200mini, X300, and N210.
https://github.com/dkozel/urh/commit/8a2b025e63fb3d30914fbe524479e2fda8c7f2b9

I'm working on a larger changeset which updates the UI to be clearer and passes around the device argument string in a variable other than `ip`, but it's not quite complete yet. It would be helpful if the actual backend in use was easier to check since `self.backend == Backends.grc` matches for a variety of devices.

dkozel, 2017-03-03:
Here's the receive side of the UI refactoring. The argument passing can probably be improved, but it is functional and I don't think it will conflict with other SDR interfaces? I don't have other hardware with me to try since I'm traveling.

@jopohl Does this look reasonable? I can do the same type of edits on the transmit side if so and open a pull request.

jopohl, 2017-03-03:
Looks like a smart solution to me @dkozel ! I would to merge a PR with this edits (and the upcoming edits for the transmit side) from you.

jopohl, 2017-03-03:
I just tested your changes with my USRP N210 and it works fine for me too! 

What would you have to enter in device args if you want to use e.g. B200? I am just asking because we may need to document this somewhere (maybe in tooltip of device args?)

dkozel, 2017-03-03:
The device arguments are all documented in the UHD manual I linked to above.
http://files.ettus.com/manual/page_identification.html

So the B200 would be `"type=b200"` or `"serial=309D05A"`, or just `""` and let UHD automatically find a USRP attached to your computer or network.

PR #187 (2017-03-12): enhance device config
- [x] Save Gains when dialog closed
- [x] remove _max_gain / _max_sample rate etc. from native device class
- [x] pass HackRF gain parameters to GR backend
- [x] hide HackRF baseband gain for Send
- [x] update HackRF IF Gain depending on send/recv
- [x] Fix high GUI gain for RTLSDR -> calc *10 for API
- [x] Add tooltips for Gain / IF Gain / Baseband Gain
- [x] use dict for saving paramaters in project manager
  - [x] save currently saved parameters
  - [x] save freq correction
- [x] fix #133 (support for USRP B200/B210)
- [x] fix #136 (HackRF Gain sliders & LNA button)
-  [... cut]
closed 2017-03-12 by jopohl