jopohl/urh #136: HackRF Gain sliders & LNA button

vsboost, 2017-02-20:
Hi there,
Not really an issue but was wondering if its possible to include gain sliders and an lna button for the hackrf?
Cheers

jopohl, 2017-02-20:
Hey @vsboost ,
thanks for bringing this up. Just some questions for clarification:
1. Are we talking about three different sliders for the HackRF?:
  - RF Gain
  - IF Gain
  - BB Gain
2. Do you know wheter the HackRF has some LNA functionality builtin? Otherwise would implement it myself. I like the idea!

vsboost, 2017-02-20:
Sorry should have called it an "RF Amp" so the three settings would be

Slider for IF LNA gain 0-40db in 8db steps
Slider for Baseband VGA gain 0-62db in 2db steps
Button for RF amp 0db = OFF ,14db = ON

Hope that makes sense

Cheers

jopohl, 2017-02-20:
Thanks for clarifying this @vsboost . I think it makes sense to have more control over these HackRF parameters, so I will implement this soon.

vsboost, 2017-02-20:
Thanks Johannes

PR #187 (2017-03-12): enhance device config
- [x] Save Gains when dialog closed
- [x] remove _max_gain / _max_sample rate etc. from native device class
- [x] pass HackRF gain parameters to GR backend
- [x] hide HackRF baseband gain for Send
- [x] update HackRF IF Gain depending on send/recv
- [x] Fix high GUI gain for RTLSDR -> calc *10 for API
- [x] Add tooltips for Gain / IF Gain / Baseband Gain
- [x] use dict for saving paramaters in project manager
  - [x] save currently saved parameters
  - [x] save freq correction
- [x] fix #133 (support for USRP B200/B210)
- [x] fix #136 (HackRF Gain sliders & LNA button)
-  [... cut]
closed 2017-03-12 by jopohl