jopohl/urh #137: Invalid argument error

mehdideveloper, 2017-02-20:
Hi
Every time I try to use one of the features of urh (like Sniffing or Spectrum Analyzer), it crashes with error number 22:

>   File "/usr/local/lib/python3.6/site-packages/urh/dev/gr/ReceiverThread.py", line 38, in run
>     self.socket.connect((self.ip, self.port))
> OSError: [Errno 22] Invalid argument
> Abort trap: 6

I've connected a RTL-SDR to my Mac OS X 10.12.2 and installed urh via pip (on Python 3.6)

Any ideas?
Thanks

jopohl, 2017-02-20:
Hi @mehdideveloper ,
I could not reproduce this behaviour, but I found [this stackeroverflow thread](http://stackoverflow.com/questions/14862742/invalid-argument-exception-in-socket-accept-if-i-restart-immediately-after-a-p). Following the solution there I called
```
setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
```
before binding the socket. Can you check if this works for you?

mehdideveloper, 2017-02-20:
Ok, thanks for the reply.
I added the following line to line 20 on SpectrumThread.py (in the "run" method, before the while loop)

`self.socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)`

Now when I try to run the spectrum analyzer, I get this error:

> Will not regenerate UI, because script cant be found. This is okay in release.
> [INFO] Stopped receiving, because no data transmitted anymore
> [INFO] Restarting with new port
> [INFO] Retry with port 51295
> QThread::start: Thread termination error: No such process
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.6/site-packages/urh/dev/gr/SpectrumThread.py", line 24, in run
>     self.socket.connect((self.ip, self.port))
> OSError: [Errno 22] Invalid argument
> Abort trap: 6

jopohl, 2017-02-20:
I think it is too late there, because the socket gets bound in the ``` init_recv_socket() ``` in AbstractBaseThread.py line 139. Did you pull latest master? I already added the call ``` self.socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1) ``` there.

Could you add a ``` print(self.ip, self.port) ``` before ``` self.socket.connect((self.ip, self.port)) ``` so we can see what is going on?

mehdideveloper, 2017-02-20:
I had installed urh using pip.
I uninstalled it, fetched the latest master and built it and now everything's fine.

Thank you

closed 2017-02-20 by mehdideveloper