jopohl/urh #138: Adding new SDRs

mehdideveloper, 2017-02-20:
Hi
How can we (contributors) add support for new SDRs (like SDRPlay and bladeRF) to urh?

jopohl, 2017-02-20:
Hi @mehdideveloper ,
glad to hear you want to add support for new SDRs! Here is a short guide to get you started:
1. add your new SDR name ``` src/urh/dev/BackendHandler.py ``` line 77 e.g. bladeRF
2. Next you have two options
  2.1 Create a __gnuradio backend__. You can look at the exisiting scripts in ``` src/urh/dev/gr/scripts ``` to get started. Make sure your script is named <lowercasename_of_your_sdr_from 1>_{send, recv} e.g. ``` bladrf_send.py ``` or ``` bladerf_recv.py ```
  2.2 Create a __native backend__ using cython. You can see an example for the hackrf in ``` src/urh/dev/native/lib/ ```. Look into ``` urh/dev/native/HackRF.py ``` to see how to use your wrapper with urh. Essentially you need to inherit from ``` urh/dev/native/Device.py ```.

It is preferred to create a native backend, if possible. However, creating a gnuradio backend requires less effort.

Let me know, if you need further assistance.

mehdideveloper, 2017-02-26:
@jopohl Are you sure these are the only changes needed? 
I have done these but I still can't see SDRPlay in my list

jopohl, 2017-02-26:
@mehdideveloper Did you implement a native or gnuradio backend? You can see an example for adding gnuradio support here: https://github.com/jopohl/urh/pull/146

mehdideveloper, 2017-02-26:
It's a GNURadio backend and I've made the two changes:
1- Added the "SDRPlay" to DEVICE_NAMES in the BackendHandler.py 
2- Created a "sdrplay_recv.grc" and "sdrplay_recv.py" in src/urh/dev/gr/scripts

Then I run the project with "python3 src/urh/main.py" and I don't see "SDRPlay" in the list of SDRs in Spectrum Analyzer or other parts.
Did I miss anything?

jopohl, 2017-02-26:
Did you enable your device in Options -> Device?

mehdideveloper, 2017-02-26:
Thanks. It solved the problem

closed 2017-02-26 by mehdideveloper