jopohl/urh #139: rtl-sdr dongle not detected

psa-jforestier, 2017-02-20:
Hello Johannes. Followinf an article published on rtl-sdr.com, I tried URH and i'm facing a dongle detection problem.
I'm using a Raspberry PI3 with a RTL-SDR USB dongle (lsusb said RTL2838 DVB-T), it works fine with all the SDR tools I use (rtl_433, SDR# ect.), but I m unable to make it work with URH.
On the Edit > Options > Device window, the line with "rtl-sdr" is always diabled and cant be unabled (checkbox is grayed).
To install URH, i followed the steps indicated on the readme.md file (I tried with the Git version of URH, and the one provided by pip3.)
How do the program decide which device is enabled or not ? Where can i got some logs on how the device is detected by URH ?

jopohl, 2017-02-20:
Hello @psa-jforestier ,
thanks for trying out URH! There are two methods for getting a device to work with URH:
1. __native backend__
2. __gnuradio backend__

Unfortunately, HackRF is currently the only device, which has a native backend. So you would need to install Gnuradio, to use the RTL-SDR.

psa-jforestier, 2017-02-21:
Ok , i will try the nuradio backend (i never uses Gnuradio, so i'm a bit anxious about that ...) . Also, i will try to add support for the native rtl-sdr device by following issue #138, cause compiling rtl-sdr tools is quite easy. I ll keep you informed.

jopohl, 2017-02-21:
Thanks @psa-jforestier, a native backend for rtl-sdr would be awesome!

psa-jforestier, 2017-02-21:
After a first look at the URH source code, i realized that i'm really not fluent in Python, and coding a Cython  wrapper to the rtl_sdr lib is out of my skills... I think coding a "rtl_tcp" wrapper will be easier : the client side (URH) need to connect via TCP to a host (can be localhost), and send command code (set frequency, set sample rate and so on) and it will receive raw IQ sample. Will have a look at this in a few days.

chibill, 2017-02-22:
Native support could be added by using pyRtlSdr.
A ctypes wrapper around librtlsdr that works on windows and linux.

Also I can't get it to detect my Windows GnuRadio install.

jopohl, 2017-02-24:
Hey @psa-jforestier  and @chibill ,
native support for rtl-sdr has been added with https://github.com/jopohl/urh/pull/154. Furthermore, the GnuRadio Issue will be fixed very soon by https://github.com/jopohl/urh/pull/147, so I am closing this one.

closed 2017-02-24 by jopohl