jopohl/urh #140: Add support for rtl_tcp "virtual" dongle

psa-jforestier, 2017-02-20:
Can you please add support for the "RTL over TCP virtual dongle" device ? it allows to stream IQ sample from a computer to an other, by using tcp.
More info here : http://osmocom.org/projects/sdr/wiki/rtl-sdr#rtl_tcp
Some SDR program already have this option (SDR# for example).
More info here : http://www.hamradioscience.com/raspberry-pi-as-remote-server-for-rtl2832u-sdr/

jopohl, 2017-02-20:
It is already possible to stream IQ data (or even binary data) from one computer to another. You simply need to activate "Options->Plugins->NetworkSDR" and then can choose "Network SDR" as new device in sending and receiving dialogs.

gustavecha, 2017-02-26:
Hi Jopohl,

I second jpforestier hint, +1 to use rtl_tcp as a particular device.
When the NetworkSDR is enabled, it's only possible to use it in protocol sniffer, not in spectrum analyzer or signal recorder. Is it the expected behaviour?

Anyway, thank you for the great work you provide to us ;)

andynoack, 2017-03-03:
Hi @gustavecha ,

that is right, NetworkSDR is designed to read and write I/Q streams from/to gnuradio. This is not a virtual SDR since there is no control channel. However, we are planing to support rtl_tcp soon.

gustavecha, 2017-03-04:
Hi @andynoack thank you for answer and good luck to get this supported :)

PR #186 (2017-03-07): Integration of rtl_tcp
fix #140
closed 2017-03-07 by jopohl