jopohl/urh #141: Installing on windows error

stevevaius77, 2017-02-21:
On windows 7 (Ultimate 64 bit), with python 3.5 (32 bit) I can not install urh via command `

> python -m pip install urh

I am receiving error
ImportError: No module named src.urh.version
What should I do to run it on windows

jopohl, 2017-02-21:
Could you try uninstalling Python 3.5 (32 bit) and install Python 3.5 (64 bit)  or Python 3.6 (64 bit) instead?

stevevaius77, 2017-02-21:
Still same problem 
[image]

stevevaius77, 2017-02-21:
Same problem still
[image]

jopohl, 2017-02-21:
I can see from your Output, that you are using python 2.7. URH requires atleast python 3.4.

stevevaius77, 2017-02-21:
Ok, I will change it now and try again :) Thnx

jopohl, 2017-02-21:
You're welcome. :)

stevevaius77, 2017-02-21:
Now it installed but dont knwo how to run because >urh command in cmd not working. Any help? :)
[image]

jopohl, 2017-02-21:
It should be found in ``` c:\users\rustu\appdata\local\programs\python36\lib\site-packages``` . If it is there, you may add this directory to your Windows Path.

stevevaius77, 2017-02-21:
Do I have to add path of \site-packages or \site-packages\urh? I added full
path bu copying (no error) of site-packages but still gives error when try
it by command >urh or >python urh

Any other help?

jopohl, 2017-02-21:
Did you find ``` urh.exe ``` in this path? If so, you should add the full path ``` c:\users\rustu\appdata\local\programs\python36\lib\site-packages ``` to your PATH variable. 

Unfortunately, I currently have no Windows for testing available right now. In some subfolder of ``` c:\users\rustu\appdata\local\programs\python36 ``` a ``` urh.exe ``` should be placed. Can you search for it? 

I will have access to my WindowsVM this evening if it still not works for you till then.

stevevaius77, 2017-02-21:
Thanks. I found the path (in scripts folder) and it loaded but still gives
error as following

[image: Inline image 1]

jopohl, 2017-02-21:
I can not see your image. :wink:

stevevaius77, 2017-02-21:
windows qt plugin error

[image]

jopohl, 2017-02-21:
That looks bad. Can you run:
```
python -c "from PyQt5 import QtWidgets"
```

stevevaius77, 2017-02-21:
I did it and after tried to commands again. Nope

jopohl, 2017-02-21:
I just want to verify, that your PyQt Installation is alrigth. Does this command give you an error:
```python 
python -c "from PyQt5.QtWidgets import QApplication; import sys; app = QApplication(sys.argv); app.exec_()"
```
You can close the terminal if no error occurs to end it.

stevevaius77, 2017-02-21:
If that was the code, here is result and still have problem
[image]

jopohl, 2017-02-21:
Yes...seems your PyQt5 installation itself is kind of broken. Can you try:
```
pip install pyqt5
```
instead
```
python -m pip install pyqt5
```
just to verify, pip is in your PATH.

stevevaius77, 2017-02-21:
[image]

jopohl, 2017-02-21:
Seems PyQt5 itself is broken here. I guess a .dll is missing or something like that. I found [this stackeroverflow thread](https://stackoverflow.com/questions/20495620/qt-5-1-1-application-failed-to-start-because-platform-plugin-windows-is-missi) suggesting to set the enviroment variable
```
QT_QPA_PLATFORM_PLUGIN_PATH=%QTDIR%\plugins\platforms\ 
```

stevevaius77, 2017-02-21:
Thanks till now, but how I can set it and I guess missing dll libegl.dll.
But again where to add this?

andynoack, 2017-02-21:
There is a small tutorial on how to set environment variables under different Windows versions: http://www.computerhope.com/issues/ch000549.htm
Have you tried something similar with the line suggested by Johannes yet?

stevevaius77, 2017-02-21:
> issues/ch000549.htm

I tried to add environment variable and adding libEGL.dll to "/platforms"
folder. No success. I have another machine which has Debian and windows
installed. I will try there too on windows

andynoack, 2017-02-21:
Maybe you can try the following with all the necessary libraries in the current path:
```urh.exe -platformpluginpath .```

stevevaius77, 2017-02-21:
Can you clarify a bit?

andynoack, 2017-02-21:
The ```-platformpluginpath .``` parameter should make urh to look for the necessary libraries in the current path. If something, e.g. libEGL.dll, cannot be found otherwise, this might help out.
However, your problem seems to be a QT5 problem. Which QT5 version do you use?

jopohl, 2017-02-21:
I just fired up a fresh Windows 7 VM and could install and run URH like a charm. @stevevaius77  maybe try to run urh from an admin terminal?

stevevaius77, 2017-02-21:
Ok I will try it Thnx

chibill, 2017-02-22:
I am able to run it. Easiest way to launch is python -m urh.main

stevevaius77, 2017-02-22:
Same result :(
"Will not regenerate UI, because script cant be found. This is okay in
release.
This application failed to start because it could not find or load the Qt
platform plugin "windows"
in "".

Reinstalling the application may fix this problem."

chibill, 2017-02-22:
Do you have pip updated and also the C++ thingy installed.

andynoack, 2017-02-23:
Hey @stevevaius77, does it work now? Can we close here?

stevevaius77, 2017-02-23:
yes the issue maybe specific to my system. I couldnt try on another windows os but no need to keep it open. Thnx

closed 2017-02-23 by andynoack