jopohl/urh #142: Mac Installation Error: ImportError: No module named 'urh.cythonext.signalFunctions'

rosstex, 2017-02-21:
The full error after running ```pip install urh``` and ```urh``` (I have XCode installed and made sure pip was anaconda's) 

```
(py35) Rosss-MacBook-Air:discussion6 rosstex$ urh
Will not regenerate UI, because script cant be found. This is okay in release.
Could not find C++ extensions, trying to build them.
Traceback (most recent call last):
  File "/Users/rosstex/anaconda/envs/py35/lib/python3.5/site-packages/urh/main.py", line 49, in main
    import urh.cythonext.signalFunctions
ImportError: No module named 'urh.cythonext.signalFunctions'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/rosstex/anaconda/envs/py35/bin/urh", line 11, in <module>
    sys.exit(main())
  File "/Users/rosstex/anaconda/envs/py35/lib/python3.5/site-packages/urh/main.py", line 55, in main
    os.chdir(os.path.join(src_dir, "urh", "cythonext"))
FileNotFoundError: [Errno 2] No such file or directory: '/Users/rosstex/anaconda/envs/src/urh/cythonext'
```

jopohl, 2017-02-21:
Seems like the exentsion modules can be built. Could you tell me what happens when you:
```
git clone htttps://github.com/jopohl/urh
cd urh
python setup.py build_ext
```

rosstex, 2017-02-21:
That seems to work fine. There's an error that "hackrf.h could not be found," but I assume that's because I don't use a hackrf radio.

jopohl, 2017-02-21:
Thats okay, this just indicates that URH will be compiled without native HackRF support. I just published a new version to PyPi so could you try again with pip?

jopohl, 2017-02-22:
Seems to be fixed.

closed 2017-02-22 by jopohl