jopohl/urh #143: works in mac

wizz777, 2017-02-21:
after install on mac dont work hackrf - all receivers in status disable

jopohl, 2017-02-21:
Hey @wizz777 , did you install a hackrf-dev package on your system? URH searches for a ``` hackrf.h ``` to compile with native HackRF support. As a fallback you may install Gnuradio and use the Gnuradio backend.

wizz777, 2017-02-21:
Hi jopohl, im try install from macport but
#sudo port install hackrf-dev
#Error: Port hackrf-dev not found
also i install gnuradio from macport and it dont see my install gnuradio(
I will be glad to help you debug on mac, i have sierra os

jopohl, 2017-02-21:
Just fired up an OSX VM and this worked for me:
```
brew update
brew upgrade
brew install hackrf 
```
after this, you need to reinstall URH, so it can find the hackrf libraries during installation.

wizz777, 2017-02-21:
thx
work but record 2-3 second and stops

log:
[INFO] HackRF-open: Success
[INFO] HackRF-set_bandwidth (1000000.0): Success
[INFO] HackRF-set_frequency (442171000.0): Success
[INFO] HackRF-set_sample_rate (1000000.0): Success
[INFO] HackRF: Starting receiving thread
[INFO] HackRF-start_rx_mode: Success
[INFO] HackRF: Stopping RX Mode: Receiving Buffer is full.
[ERROR] HackRF: Could not join read_queue_thread
[INFO] stopping HackRF rx mode (Receiving Buffer is full.)
[WARNING] closing because stop_rx_mode of HackRF is bugged and will not allow re receive without close
[INFO] HackRF: Attempting to close...
[INFO] Successfully closed HackRF
[INFO] HackRF-close: Success
[INFO] HackRF: Stopping RX Mode: Stopped receiving: Stop button clicked

jopohl, 2017-02-21:
``` [INFO] HackRF: Stopping RX Mode: Receiving Buffer is full. ``` indicates, that there is no space left to save the recorded signal. How much RAM you got in your machine?

wizz777, 2017-02-21:
8Gb

jopohl, 2017-02-21:
Thats really weird...URH uses up to 60% of available RAM when recording. Can you run 
```python3 -c "import psutil; print(0.6*(psutil.virtual_memory().free / 8)) ``` and print the result?

wizz777, 2017-02-21:
#python3 -c "import psutil; print(0.6*(psutil.virtual_memory().free / 8))" 
30883430.4

wizz777, 2017-02-21:
why you dont write from memory on disk? btw

jopohl, 2017-02-21:
Spare Life of SSDs. :wink: I guess this is some MAC OS special behaviour as I already recorded signals for minutes on 8gb RAM machines. However, I could implement some kind of fallback to write on disk, if RAM runs empty.

wizz777, 2017-02-21:
soon will come my usrp, be ready) couse i dont see enable usrp in urh ;)

jopohl, 2017-02-21:
Unfortunately, there is no native support for USRP in URH, yet. I am working on it! In the meantime you may use the gnuradio backend.

wizz777, 2017-02-21:
in any case, thank you for your hard work is the coolest program for me

jopohl, 2017-02-21:
That's great to hear. Thanks @wizz777 !

closed 2017-02-22 by jopohl