jopohl/urh #144: "Bit length" vs. "symbol length"

OZ1SEJ, 2017-02-21:
It strikes me as odd that the smallest temporal unit in a signal is referred to as a "bit" when technically it's a symbol. Unless the modulation index is 2 (e.g. BPSK or OOK), there may be more than one bit represented by one signal state.

andynoack, 2017-02-22:
Well, that's true. We had the this discussion several weeks ago and decided to go with bit length for simplicity reasons as currently no modulations with index >=2 are supported. 
We haven't found any attackable wireless protocol in the wild (i.e. IoT, Smarthome, keyfobs, ...) that operates with more information per symbol. Therefore it would be very great if you could provide us with some non-binary signals. Handling them will (probably) be a feature of the next major version of URH. And I promise, we will have technically sound notations then ;-)

OZ1SEJ, 2017-02-22:
Thank you for your kind answer! I thought I had misunderstood something ;-)

closed 2017-02-22 by OZ1SEJ