jopohl/urh #145: Does not recognize Windows Install of GNURadio.

chibill, 2017-02-22:
No matter what I do I can't get it to see my GNURadio install. (Which was installed with GNURadio's Windows installer)

jopohl, 2017-02-22:
Thanks for bringing this up @chibill .
I made a nearly working fix in https://github.com/jopohl/urh/pull/147.

However, Gnuradio itself seems to be kind of bugged on Windows. The Gnuradio Backend is based on TCP sockets like this:
[image]

However, if I try to connect via python to this socket with:
```python
import socket
socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
socket.connect(("127.0.0.1", 1337))
```
Gnuradio just exits completly. On Linux this works like a charm.

Any ideas on this?

chibill, 2017-02-22:
No idea. So the gnuradio back end is basiclly a rtl_tcp backend. Can you at least make it so we can just start rtl_tcp and use that?

Also you could directly use the librtlsdr that already has a ctypes wrapper for it.

chibill, 2017-02-22:
I can try what you posted on my end and see if anything happens.

jopohl, 2017-02-22:
Let me know, if you find out anything.

chibill, 2017-02-22:
Also I does not like where I point it to my python 2.7.13 install.

Also the TCP Sink is deprecated so that could cause  a problem later on.

No idea why it hangs it but. Makes python go none responsive.

Anyway for me to make urh connect to rtl_tcp server instead?

jopohl, 2017-02-22:
In perspective we will move away from Gnuradio backend. First, the TCP sink is deprecated as you mentioned and second, the UDP sink also does not work. Hang on, we are currently working on enhancing the native device support.

For ``` rtl_tcp``` you could have a look at ``` urh/plugins/NetworkSDRInterface/NetworkSDRInterfacePlugin.py ``` and implement a  ``` SDRPlugin ```. This is intended for this kind of stuff.

chibill, 2017-02-22:
There is also using librtlsdr directly. (Or using pyrtlsdr a ctypes wrapped around it)

jopohl, 2017-02-22:
Yes, creating a librtlsdr wrapper with Cython has high priority for me currently.

chibill, 2017-02-22:
You could look at the pure python ctypes wrapper for help with that maybe?

jopohl, 2017-02-22:
Yes that will be my starting point. :wink:

andynoack, 2017-02-23:
Gnuradio has a confirmed bug on Windows in respect to the TCP sink we are using for the SDR-backend: https://github.com/gnieboer/GNURadio_Windows_Build_Scripts/issues/15
We are working on a solution! Stay tuned.

jopohl, 2017-02-25:
Fixed with https://github.com/jopohl/urh/pull/147

closed 2017-02-25 by jopohl