jopohl/urh #156: Crash when click analyze in the analysis tab

vsboost, 2017-02-25:
Was playing with an ASK signal and program exits when i clicked the analyze button.

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/urh/controller/CompareFrameController.py", line 1043, in on_btn_analyze_clicked
    self.proto_analyzer.auto_assign_labels()
  File "/usr/local/lib/python3.5/dist-packages/urh/signalprocessing/ProtocolAnalyzer.py", line 842, in auto_assign_labels
    format_finder = FormatFinder(self)
  File "/usr/local/lib/python3.5/dist-packages/urh/awre/FormatFinder.py", line 29, in __init__
    self.bitvectors = [np.array(msg.decoded_bits, dtype=np.int8) for msg in self.protocol.messages]
  File "/usr/local/lib/python3.5/dist-packages/urh/awre/FormatFinder.py", line 29, in <listcomp>
    self.bitvectors = [np.array(msg.decoded_bits, dtype=np.int8) for msg in self.protocol.messages]
TypeError: int() argument must be a string, a bytes-like object or a number, not 'src.urh.cythonext.signalFunctions.Symbol'
Aborted

jopohl, 2017-02-25:
Thanks for reporting this @vsboost . I will have a look at this.

BTW: The analyze button is currently in a beta state. We are planning to supercharge it with some artificial intelligence in the future, so keep an eye on it. :wink:

closed 2017-02-25 by jopohl

vsboost, 2017-02-25:
No probs :-)

commit 35257f1a (2017-02-25): fix #156