jopohl/urh #159: Unresolved external symbol (RTLSDR)

cimba007, 2017-02-25:
[output.txt](https://github.com/jopohl/urh/files/801111/output.txt)

Before the latest merge of fix_windows_native it was at least building .. now I got missing external symbol. What am I missing?

I have installed gnuradio from the windows-installer.

jopohl, 2017-02-25:
Seems to me, you are using a 32 Bit version of python. We only support the 64 Bit version.

cimba007, 2017-02-25:
Thanks, would be easy if this is the point ;_)

jopohl, 2017-02-25:
Glad the native backend is working for you now. :+1: 

Gnuradio backend for windows will be fixed in: https://github.com/jopohl/urh/pull/147, which is nearly done. New release should be ready today or tomorrow.

cimba007, 2017-02-25:
Very nice work you guys put up there. Keep it going!!!

closed 2017-02-25 by cimba007