jopohl/urh #160: More info on wav file import

vikbez, 2017-02-25:
Can you add more info on wav file import ?
I tried many different formats but could not get it to work properly

andynoack, 2017-02-25:
Hey @vikbez,
try the following Wave-Format: RIFF (little-endian) data, WAVE audio, Microsoft PCM, 8 bit, mono.
Furthermore you have to know whether your wave-files are quadrature demulated or not. You can configure this in the checkbox popping up when loading a file via File->Open...

cimba007, 2017-02-25:
After I got some problems on capturing signals with urh ( https://github.com/jopohl/urh/issues/161 ) I tried to record some audio in sdr-sharp an tried to import it.

None of the two 8-bit-PCM files are working. Am I missing something huge here?

[image]

[image]

[image]

andynoack, 2017-02-25:
Would you upload one of your recorded wave-files? This would help us analyze the problem very much!

cimba007, 2017-02-25:
I uploaded both files:

https://mega.nz/#F!Pw4RGCJR!35-l9vpcGBrFbHZjIUAxgA

The AF-Files contains the audio-signal as AM-Modulation:

[image]

andynoack, 2017-02-25:
Thank you, I will have a look on it!

cimba007, 2017-02-25:
Problem might be on my end .. apparently it is not correctly exported to 8bit-mono .. trying to convert the sdr-sharp file afterwards.

andynoack, 2017-02-25:
Great! Can we close this issue?

cimba007, 2017-02-25:
I did't open it but for now I am totally fine with the workaround with audacity.

closed 2017-02-25 by andynoack

vikbez, 2017-02-25:
👍  thanks, works with 8bit-mono

nofxmrngstr11, 2023-02-17:
hello im having trouble saving or even opening any recorded signals in urh. everytime i attempt to save a recorded signal the app just closes out. then im left with this error in my terminal:

"/snap/urh/9/usr/lib/python3.6/multiprocessing/semaphore_tracker.py:143: UserWarning: semaphore_tracker: There appear to be 2 leaked semaphores to clean up at shutdown
  len(cache))"

im using parrot sec os btw. any suggestions or possible solutions?

andynoack, 2023-02-17:
This is actually not the right place for a new issue. As you can see in your stack trace the error occurs in the python libs. If you open a new issue on that, ensure that the place where URH produces the error is within the stacktrace. I assume that this is a library/dependency bug not specific to URH, however.