jopohl/urh #161: Capturing on Windows to slow? (RTLSDR)

cimba007, 2017-02-25:
Hello,

after watching this video (https://www.youtube.com/watch?v=kuubkTDAxwA) I got really excited about urh ...

But somehow my capture-rate is very low .. analyzer takes 100% cpu .. somehow it just doesn't feel right. Capturing a signal seems to be very slow .. I only got 0,2seconds captured after like 10seconds realtime .. don't know what is the meaning of this.

[image]

jopohl, 2017-02-25:
Hey @cimba007 ,

thanks for the quick tryout of the new feature!

Indeed there was a bug, with a too high timeout, so the recording time did elapse so slow. That was an easy fix.

However, things are getting a bit more complicated on Windows, as I can not use Processes there, because Python Processes cannot share ressources on Windows: http://rhodesmill.org/brandon/2010/python-multiprocessing-linux-windows/

That's why I need to use Threads on Windows which leads to the CPU problems you experience. Some further research is required here.

cimba007, 2017-02-25:
@jopohl: gnuradio-backend is working fine. 
 Good work!

jopohl, 2017-02-25:
Thanks @cimba007 . I am closing this one. Optimization of native RTL-SDR backend will be tracked here: https://github.com/jopohl/urh/issues/164

closed 2017-02-25 by jopohl