jopohl/urh #162: Improve File import from SDRSharp

OmorzzMo, 2017-02-25:
I have an RTL-SDR USB dongle, but the built-in recordings are not usable: level almost zero.

So I open up SDRSharp (SDR# 1.0.0.1500) where I can see my signal beautifully. I can record the data nicely and look at it with Audacity. However, none of the possible export file formats, i.e. WAV with 8-bit PCM, 16-bit PCM oder 32 bit IEEE float, Baseband-Recording can be imported into URH. These files are simple stereo audio files with IQ-values.
The importer throws an error message like unsupported format.

andynoack, 2017-02-25:
We have recently made some changes to the code. Aside from the SDR# import, you may try to record with URH in the latest git version.

OmorzzMo, 2017-02-25:
I tired the version 1.4

By the way, on method described under Running from source->"Without Installation" doesn't work on my Windows 10 system, because when I try to execute the source from src/urh :> with Python 3.6.0 |Anaconda 4.3.0 (64-bit)|  I get an error:

urh/src/urh>python main.phy

Traceback (most recent call last):
  File "main.py", line 119, in <module>
    main()
  File "main.py", line 21, in main
    assert os.path.isdir(urh_dir)
AssertionError

The reason is, that os.path.dirname(__file__) is empty in this case.
You have to call it always with the full path, like pythond C:\Dir1\Dir2\uhr\src\uhr\main.py 
Then it will work.

jopohl, 2017-02-25:
Hi @OmorzzMo 

thanks for pointing me at this. I fixed the Assertion error with the path name in 4fc16844fff377f1b17f57436b2450b939c275f6 so you can run from inside urh dir now.

BTW: Did you try the workaround for SDR# mentioned in: https://github.com/jopohl/urh/issues/160 ?

closed 2017-03-15 by andynoack