jopohl/urh #167: Unable to install latest update.

chibill, 2017-02-27:
> Requirement already satisfied: numpy in c:\python35\lib\site-packages (from urh)
Requirement already satisfied: psutil in c:\python35\lib\site-packages (from urh)
Requirement already satisfied: pyzmq in c:\python35\lib\site-packages (from urh)
Installing collected packages: urh
  Running setup.py install for urh ... error
    Complete output from command C:\Python35\python3.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\chibi\\AppData\\Local\\Temp\\pip-build-2p3hrxnv\\urh\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\chibi\AppData\Local\Temp\pip-1qupfc9k-record\install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build\lib.win32-3.5
    creating build\lib.win32-3.5\urh
    copying src\urh\constants.py -> build\lib.win32-3.5\urh
    copying src\urh\FFTSceneManager.py -> build\lib.win32-3.5\urh
    copying src\urh\LiveSceneManager.py -> build\lib.win32-3.5\urh
    copying src\urh\main.py -> build\lib.win32-3.5\urh
    copying src\urh\SceneManager.py -> build\lib.win32-3.5\urh
    copying src\urh\SignalSceneManager.py -> build\lib.win32-3.5\urh
    copying src\urh\version.py -> build\lib.win32-3.5\urh
    copying src\urh\__init__.py -> build\lib.win32-3.5\urh
    creating build\lib.win32-3.5\urh\awre
    copying src\urh\awre\CommonRange.py -> build\lib.win32-3.5\urh\awre
    copying src\urh\awre\FormatFinder.py -> build\lib.win32-3.5\urh\awre
    copying src\urh\awre\__init__.py -> build\lib.win32-3.5\urh\awre
    creating build\lib.win32-3.5\urh\awre\components
    copying src\urh\awre\components\Address.py -> build\lib.win32-3.5\urh\awre\components
    copying src\urh\awre\components\Component.py -> build\lib.win32-3.5\urh\awre\components
    copying src\urh\awre\components\Flags.py -> build\lib.win32-3.5\urh\awre\components
    copying src\urh\awre\components\Length.py -> build\lib.win32-3.5\urh\awre\components
    copying src\urh\awre\components\Preamble.py -> build\lib.win32-3.5\urh\awre\components
    copying src\urh\awre\components\SequenceNumber.py -> build\lib.win32-3.5\urh\awre\components
    copying src\urh\awre\components\Type.py -> build\lib.win32-3.5\urh\awre\components
    copying src\urh\awre\components\__init__.py -> build\lib.win32-3.5\urh\awre\components
    creating build\lib.win32-3.5\urh\controller
    copying src\urh\controller\CompareFrameController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\DecoderWidgetController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\FuzzingDialogController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\GeneratorTabController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\MainController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\MessageTypeDialogController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\ModulatorDialogController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\OptionsController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\PluginController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\ProjectDialogController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\ProtocolLabelController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\ProtocolSniffDialogController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\ReceiveDialogController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\SendDialogController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\SendRecvDialogController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\SignalDetailsController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\SignalFrameController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\SignalTabController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\SpectrumDialogController.py -> build\lib.win32-3.5\urh\controller
    copying src\urh\controller\__init__.py -> build\lib.win32-3.5\urh\controller
    creating build\lib.win32-3.5\urh\cythonext
    copying src\urh\cythonext\analyze.py -> build\lib.win32-3.5\urh\cythonext
    copying src\urh\cythonext\build.py -> build\lib.win32-3.5\urh\cythonext
    copying src\urh\cythonext\__init__.py -> build\lib.win32-3.5\urh\cythonext
    creating build\lib.win32-3.5\urh\dev
    copying src\urh\dev\BackendHandler.py -> build\lib.win32-3.5\urh\dev
    copying src\urh\dev\PCAP.py -> build\lib.win32-3.5\urh\dev
    copying src\urh\dev\VirtualDevice.py -> build\lib.win32-3.5\urh\dev
    copying src\urh\dev\__init__.py -> build\lib.win32-3.5\urh\dev
    creating build\lib.win32-3.5\urh\dev\gr
    copying src\urh\dev\gr\AbstractBaseThread.py -> build\lib.win32-3.5\urh\dev\gr
    copying src\urh\dev\gr\ReceiverThread.py -> build\lib.win32-3.5\urh\dev\gr
    copying src\urh\dev\gr\SenderThread.py -> build\lib.win32-3.5\urh\dev\gr
    copying src\urh\dev\gr\SpectrumThread.py -> build\lib.win32-3.5\urh\dev\gr
    copying src\urh\dev\gr\__init__.py -> build\lib.win32-3.5\urh\dev\gr
    creating build\lib.win32-3.5\urh\dev\gr\scripts
[... 151 lines cut ...]
    copying src\urh\util\Logger.py -> build\lib.win32-3.5\urh\util
    copying src\urh\util\ProjectManager.py -> build\lib.win32-3.5\urh\util
    copying src\urh\util\__init__.py -> build\lib.win32-3.5\urh\util
    package init file 'src\urh\__pycache__\__init__.py' not found (or not a regular file)
    copying src\urh\cythonext\path_creator.cpp -> build\lib.win32-3.5\urh\cythonext
    copying src\urh\cythonext\signalFunctions.cpp -> build\lib.win32-3.5\urh\cythonext
    copying src\urh\cythonext\util.cpp -> build\lib.win32-3.5\urh\cythonext
    copying src\urh\cythonext\path_creator.pyx -> build\lib.win32-3.5\urh\cythonext
    copying src\urh\cythonext\signalFunctions.pyx -> build\lib.win32-3.5\urh\cythonext
    copying src\urh\cythonext\util.pyx -> build\lib.win32-3.5\urh\cythonext
    copying src\urh\dev\native\lib\hackrf.cpp -> build\lib.win32-3.5\urh\dev\native\lib
    copying src\urh\dev\native\lib\rtlsdr.cpp -> build\lib.win32-3.5\urh\dev\native\lib
    copying src\urh\dev\native\lib\hackrf.pyx -> build\lib.win32-3.5\urh\dev\native\lib
    copying src\urh\dev\native\lib\rtlsdr.pyx -> build\lib.win32-3.5\urh\dev\native\lib
    copying src\urh\dev\native\lib\chackrf.pxd -> build\lib.win32-3.5\urh\dev\native\lib
    copying src\urh\dev\native\lib\crtlsdr.pxd -> build\lib.win32-3.5\urh\dev\native\lib
    creating build\lib.win32-3.5\urh\dev\native\lib\win
    copying src\urh\dev\native\lib\win\COPYING.libusbx -> build\lib.win32-3.5\urh\dev\native\lib\win
    copying src\urh\dev\native\lib\win\COPYING.pthreads-win32 -> build\lib.win32-3.5\urh\dev\native\lib\win
    copying src\urh\dev\native\lib\win\hackrf.dll -> build\lib.win32-3.5\urh\dev\native\lib\win
    copying src\urh\dev\native\lib\win\hackrf.lib -> build\lib.win32-3.5\urh\dev\native\lib\win
    error: can't copy 'src\urh\dev\native\lib\win\libhackrf': doesn't exist or not a regular file

    ----------------------------------------
Command "C:\Python35\python3.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\chibi\\AppData\\Local\\Temp\\pip-build-2p3hrxnv\\urh\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\chibi\AppData\Local\Temp\pip-1qupfc9k-record\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in C:\Users\chibi\AppData\Local\Temp\pip-build-2p3hrxnv\urh\

chibill, 2017-02-27:
Do I have to have a 64 bit version of Py3 to install and use urh??? After the recent update to get the rtl_sdr wroking?

jopohl, 2017-02-27:
Hey @chibill ,

yes you need to have a 64 Bit version of Python 3.

chibill, 2017-02-27:
When did this become a requirement? Like really...

Also why use 64 but when many of the binary based libs don't have 64 bit
versions example scipy.

But what ever I will see if I can get it to install with out wiping out my
py2... Stupid py3 loves to do that on Windows...

<https://github.com/jopohl/urh/issues/167#issuecomment-282641181>, or mute
the thread

jopohl, 2017-02-27:
I see your point @chibill . I added support for python 3 32 bit again in latest release. Native backends will be disabled and an message will be shown in Device options to indicate that.

chibill, 2017-02-27:
I am confused why the native back end would not work with 32 bit.and least
I know the rtl_SDR driver has both versions.. (And most python based ones
are 32 bit. Like the one pyrtlsdr uses.

jopohl, 2017-02-27:
It is just a matter of resources. I simply do not have the time to maintain and test native backends with 32 bit python 3. I thought anyone using this kind of software would use 64 bit python 3 anyway, as you need a couple of  usable RAM to work properly with signals.

That said, if you would like helping increasing the windows support and specifically supporting 32 bit python 3 for native backend, be welcome.

chibill, 2017-02-27:
The thing is I only had 32 installed because that's what everyone required
at the time for native C stuff. (I have 3.5 installed and I live on py2
because it's stable and also integers works correctly.)

Like I said I will try to get py3 64 working. Just don't want to break tons
of stuff... (Like ipython.)

jopohl, 2017-02-27:
Time goes on and so do dependencies. :wink: 

Furthermore, it sounds to me that you can go on with your exisiting programs, as you are running them with python 2 (32 bit), so this should be untouched by an python 3 (64 bit) installation.

I will close this one. If other problems should arise during installation feel free to open another issue.

closed 2017-02-27 by jopohl

chibill, 2017-02-27:
Okay. Hopefully this works then..