jopohl/urh #168: TCP or Zmq sink

mehdideveloper, 2017-02-27:
Older GNURadio backends or those added by other developers (like bladeRF) use TCP Sink, but you've disabled TCP sink in your own .grc files and have added Zeromq sink. 
What do you recommend for the new backends? What are the differences in performance?

andynoack, 2017-02-27:
Hi @mehdideveloper 
the problem is that Gnuradios TCP sink is deprecated and furthermore not running on Windows. Therefore we recently searched for an alternative and this is ZeroMQ. However, I would prefer a conventional socket solution but then we had to drop Windows support...

mehdideveloper, 2017-02-27:
Hi @andynoack 
Thanks for the prompt reply.
Well I didn't notice that you're also supporting Windows.

closed 2017-02-27 by mehdideveloper