jopohl/urh #172: Incompatible with python2 or gnuradio not installed (macos)

KenMurphy, 2017-02-28:
I installed URH via pip3. When I run it, in the options menu it says that gnuradio is not installed or incompatible with the python2 interpreter. It finds my python2 interpreter fine as shown in the window. I installed gnuradio via macports before running urh.

jopohl, 2017-02-28:
What does ``` python2 -c "import gnuradio" ``` say?

KenMurphy, 2017-02-28:
ModuleNotFoundError: No module named 'gnuradio'

So it appears to be more of a python/gnuradio problem I guess. Sorry, you
can close this issue.

jopohl, 2017-02-28:
No problem. Maybe you can use a native backend for capturing, so there is no need for using gnuradio? We are currently enhancing our native support, so stay tuned!

closed 2017-02-28 by jopohl

KenMurphy, 2017-02-28:
Hmm, I'll play around with it and see what happens. And will do, excited to
try this software out!

KenMurphy, 2017-03-02:
Just in case anybody comes across this issue looking for a solution:

I fixed it by updating macports and then specifying the python version from ports manually. In my case I entered '/opt/local/bin/python2.7' into the 'Python2 interpreter' box in URH and hit enter. It seems to work now :)