jopohl/urh #173: Can't enable device in macOS 10.12.2

matix2120, 2017-02-28:
I've tried to install urh using pip3 and also build from sources. In each case I was not able to enable rtlsdr in settings (this option is grayed out). librtlsdr is installed. Device is physically connected to the usb and works fine in gqrx or cubicsdr.

Log from the compilation:
http://pastebin.com/ZPWTC9zu

jopohl, 2017-02-28:
Line 25: ``` ld: library not found for -lrtlsdr ``` indicates, that it can not find rtlsdr library. What does 
```
sudo updatedb
locate rtlsdr.so
```
say?

matix2120, 2017-03-01:
locate rtlsdr:
http://pastebin.com/jSAHJiBf

> $ ld -v -lrtlsdr
@(#)PROGRAM:ld  PROJECT:ld64-274.2
configured to support archs: armv6 armv7 armv7s arm64 i386 x86_64 x86_64h armv6m armv7k armv7m armv7em (tvOS)
Library search paths:
	/usr/lib
	/usr/local/lib
Framework search paths:
	/Library/Frameworks/
	/System/Library/Frameworks/
ld: warning: -arch not specified
ld: warning: -macosx_version_min not specified, assuming 10.10
Undefined symbols for architecture x86_64:
  "start", referenced from:
     implicit entry/start for main executable
ld: symbol(s) not found for inferred architecture x86_64

And what is more:

> $ brew install librtlsdr
Warning: librtlsdr-0.5.3 already installed

So looks like librtlsdr is installed, but clang still can't find it. Can you help me with adding it correctly?

jopohl, 2017-03-01:
Mhh looks fine to me...can you run:
```
git clone https://github.com/jopohl/urh
python3 ./src/urh/cythonext/build.py
```
and see if it finds your rtlsdr lib? If so, you can reinstall URH using:
```
pip3 uninstall urh
pip3 --no-cache-dir install urh
```

matix2120, 2017-03-01:
I've built it as you said, but it looks the same like when I was running main.py:
`ld: library not found for -lrtlsdr`
`clang: error: linker command failed with exit code 1 (use -v to see invocation)`

Uninstalling and installing again with `--no-cache-dir` also didn't change anything.

jopohl, 2017-03-01:
What does 
```
echo $PATH
```
give you?

matix2120, 2017-03-01:
`$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/usr/local/CrossPack-AVR/bin`

jopohl, 2017-03-01:
I am not a real OS X pro, but according to this discussion [on stackoverflow](http://stackoverflow.com/questions/3146274/is-it-ok-to-use-dyld-library-path-on-mac-os-x-and-whats-the-dynamic-library-s) you could try setting to DYLD_LIBRARY_PATH to "/usr/local/lib/".

matix2120, 2017-03-01:
Unfortunately that didn't help :(

jopohl, 2017-03-01:
Can you put this little c program:
```c
main (int argc, char **argv) {
    rtlsdr_get_device_name();
}
```
somewhere and call ``` clang -lrtlsdr <yourfile>.c ``` and see if it works? This is the check URH performs during installing to see, if rtlsdr is available.

matix2120, 2017-03-01:
> ld: library not found for -lrtlsdr
> clang: error: linker command failed with exit code 1 (use -v to see invocation)

jopohl, 2017-03-01:
Well there we got it...somehow your clang does not find it. What does:
```
clang -lrtlsdr -L/usr/local/lib <yourfile>.c
```
give?

matix2120, 2017-03-01:
Looks better. 

> $ clang -lrtlsdr -L/usr/local/lib test.c
test.c:1:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
main (int argc, char **argv) {
^
test.c:2:5: warning: implicit declaration of function 'rtlsdr_get_device_name' is invalid in C99 [-Wimplicit-function-declaration]
    rtlsdr_get_device_name();
    ^
2 warnings generated.

How to tell clang to look in /usr/local/lib by default?

matix2120, 2017-03-01:
According to this [stackoverflow thread](http://stackoverflow.com/questions/23905661/on-mac-g-clang-fails-to-search-usr-local-include-and-usr-local-lib-by-def) I needed to install xcode command line tools:
`xcode-select --install` and now clang is able to find librtlsdr.

andynoack, 2017-03-01:
Only the latest rtlsdr driver supports bandwidth control, the older versions just do not have the symbol that is missing here. However, we have implemented a fallback module that lets you use rtlsdr with an old driver. If you wish to control bandwidth, install the latest version of librtlsdr on your system and reinstall urh/rebuild the modules.

matix2120, 2017-03-01:
Where is the proper source of that driver?

andynoack, 2017-03-01:
You should get it from here: http://osmocom.org/projects/sdr/wiki/rtl-sdr

jopohl, 2017-03-01:
Very interesting @matix2120 ! I created a little fix for that in ``` setup.py ``` with 2d7dd0eff687fdb1cb3ab6c576708cb8b15a69ba, so other users do not need to install xcode to enjoy native backends. Thanks for your help!

matix2120, 2017-03-01:
I'm glad I could help you, and thank you very much for your help. 
I've compiled the rtl-sdr lib and now urh seems working. I've got only one not related question, is it normal that in Record signal window only 1 114 112 samples are captured?

jopohl, 2017-03-02:
Glad it works for you now!
The maximum number of recorded samples depends on your available RAM at the time you press the record button. Currently, 60% of available RAM are reserved for the receive buffer. This threshold will be made configurable in options in https://github.com/jopohl/urh/issues/148.

matix2120, 2017-03-02:
Ok, great. Thank you very much for your help. I think we can mark the issue as closed and resolved.

jopohl, 2017-03-02:
You're welcome @matix2120 .

closed 2017-03-02 by jopohl