jopohl/urh #174: Add support for LimeSDR

gerryk, 2017-02-28:
This tool would really improve the usability of LimeSDR.
SoapySDR or GR might be a good start.

andynoack, 2017-02-28:
We are actually planning to support LimeSDR soon, stay tuned!

n1jez73, 2017-04-12:
I installed and updated under Windows and the programs runs, but the LimeSDR is not in the devices list....

I did do an update just now.

andynoack, 2017-04-12:
LimeSDR is not yet supported, but will be in the next days when our tests have been completed. You can try the limesdr branch for beta testing if you like!

n1jez73, 2017-04-12:
Ok, great! looking forward to it. 

---

73,
Mike, N1JEZ
"A closed mouth gathers no feet" 

 

Links:
------
[1] https://github.com/jopohl/urh/issues/174#issuecomment-293730799
[2]
https://github.com/

closed 2017-04-14 by jopohl

n1jez73, 2017-04-14:
> Closed #174 <https://github.com/jopohl/urh/issues/174> via #228

updated and working!

Tnx
Mike

PR #228 (2017-04-14): Add support for limesdr
This adds support for LimeSDR to fix #174 

- [x] wrap save method
- [x] Cython Wrapper + Test methods for each
   - [x] Sample Rate
   - [x] Gain
   - [x] Center Freq
   - [x] Bandwidth
   - [x] Antenna
   - [x] Temperature
   - [x] RX Stream
   - [x] TX Stream
- [x] adapt config.py for LimeSDR limits
- [x] Add Python Class
  - [x] refactor ``` process_command ``` 
     - [x] use enum for commands
     - [x] use generic ref for native lib in base class
     - [x] make method names configurable via defaultdict, if None/Unspupported -> Do nothing
  - [x] Mehod for receiv [... cut]