jopohl/urh #175: Deps for rtlsdr and hackrf appear to be automagic

ZeroChaos-, 2017-03-01:
librtlsdr and libhackrf are not listed as deps, however, they are linked to.  It appears the linking may be conditional on finding the libs, but my knowledge of distutils is pretty weak.

Is there a flag to force these deps on or off?  If not, can there please be?  I'm adding this to gentoo with a hard dep on rtlsdr and hackrf but I'd much rather make them conditional.

BTW, thanks for your clear and concise directions, your readme, your git tags, and your youtube videos.  It's rare I run into SDR related projects that are so "together".  Thanks for all your hard work, both on the software and all the things that make my life (and your users lives) better.

andynoack, 2017-03-01:
That is pretty easy (@jopohl correct me, if I am wrong). If the libs are in place, they are built in at the first run of URH or when you run cythonext/build.py manually. Therefore both libs are no hard deps but USE-flags (without changing anything on URH - just install the requirements) would be very nice!

ZeroChaos-, 2017-03-01:
Unfortunately without changing URH I can't make them use flags.  If a user has rtlsdr installed but for whatever reason sets USE="" (empty) for urh, the use flags wouldn't be respected.  I need a way in urh to disable/enable a dep instead of it being automagic.

andynoack, 2017-03-01:
I see. Should be possible somehow, I will discuss this with @jopohl

jopohl, 2017-03-01:
I just pushed a suggestion to branch __explicit_deps__ in 808308a2ac808d0d778b5135dc6e57fb807fcdaa. With this, you can enforce ignoring the device extensions with
```
 python setup.py install --without-hackrf --without-rtlsdr
```
or enforce installing them  with:
```
 python setup.py install --with-hackrf --with-rtlsdr
```
Of course, mixtures are also possible:
```
 python setup.py install --without-hackrf --with-rtlsdr
```
If you do not give a flag for a device, the previous automagic will be applied. Will this work for you @ZeroChaos- ?

ZeroChaos-, 2017-03-01:
I'm terrible at reading this distutils stuff so just to confirm, what happens if I pass "--with-hackrf" and libhackrf cannot be found, and what happens if I pass --without-hackrf and libhackrf can be found?

jopohl, 2017-03-02:
Well, if you pass ``` --without-hackrf ``` and libhackrf can be found you will not have native support for HackRF in URH, no matter if it is installed. It will just skip the Extension in any case.
On the other hand, if you give ``` --with-hackrf  ``` and libhackrf can __not__ be found, URH will fail to build.

jopohl, 2017-03-02:
I merged these changes into master now.

ZeroChaos-, 2017-03-02:
awesome, this sounds like exactly what I wanted. thank you very much.  I'll update in gentoo after your next tag.

ZeroChaos-, 2017-03-02:
lol, which apparently right now :-D

closed 2017-03-02 by ZeroChaos-