jopohl/urh #176: bladerf support

ZeroChaos-, 2017-03-01:
bladerf support as well please.  directly of pull in a bunch of hardware using gr-osmosdr and/or soapysdr support.

ZeroChaos-, 2017-03-01:
after using for more than 10 seconds, it appears this is already supported.  I suppose my question is "how", so that I can add the appropriate deps (if any) for other users.

andynoack, 2017-03-01:
Currently BladeRF is only supported via the gnuradio backend. That means, gnuradio + gr-osmosdr should be sufficient to use it.

ZeroChaos-, 2017-03-01:
Does that mean that gr-osmosdr is a (potential) dep?

andynoack, 2017-03-01:
Yes it is. Both gnuradio and gr-osmosdr are necessary to enable the gnuradio backend. At the moment I am not sure what happens, if gnuradio is installed without gr-osmosdr...

ZeroChaos-, 2017-03-01:
Cool, I've updated gentoo and I'll leave this open as a reminder to please update your readme (and anywhere else you feel appropriate) that gr-osmosdr is at least an optional dep.  Might want to specify libhackrf and librtlsdr in the same place.

closed 2017-03-02 by jopohl

ZeroChaos-, 2017-03-02:
awesome, thanks!

commit c13c4860 (2017-03-02): fix #176