jopohl/urh #177: RTL-SDR Low Signal Strength on Record Signal

cspain, 2017-03-01:
Windows 10 using NooElec RTL-SDR USB Dongle:

Signal when using the spectrum analyzer/tuner is good, but when recording the signal is very weak - sometimes barely above the noise despite the gain being turned up.

This problem was briefly mentioned in Issue #162.

Thank you.

andynoack, 2017-03-01:
Confirmed. We will search for the bug!

jopohl, 2017-03-01:
@cspain what backend are you using? Native Backend for RTL-SDR works fine on Windows 10 for me.

cspain, 2017-03-01:
I have attached some screen grabs with the issue. Maybe the noise floor is
too high.

Thanks!

jopohl, 2017-03-01:
@cspain I can not see your screen grabs. Maybe try putting the gain to 300 like @andynoack mentioned?

cspain, 2017-03-01:
Putting the gain to 300 worked. Thank you.

This tool is great by the way and I appreciate all the work put into it and
the fact you're sharing it.

jopohl, 2017-03-01:
You're welcome @cspain . Glad it works now!

closed 2017-03-01 by jopohl

gustavecha, 2017-03-01:
Hi

I've also had this problem last sunday during my tests

RTL-SDR devices' gain is limited to 40, how would you do to set it to 300 ?
Why spectrum analyser works well without this 300 gain ?
It seems pretty unrealistic to set gain to 300, it's a really high value

By the way, as cspain said, the works done here is huge and highly appreciated, big thank you !

cspain, 2017-03-01:
I was limited to 40 gain in the past (maybe an earlier build) , when I
tried it again today I was able to set 300.

jopohl, 2017-03-01:
HI @gustavecha ,

as @cspain correctly assumed it was indeed a buggy limitation in an earlier version to restrict RTL-SDR to gain 40. RTL-SDR API demands such high values, there is nothing much I can do about it.

The whole gain setting will be made clearer with implementation of: https://github.com/jopohl/urh/issues/148. I will add a slider for the gain, so it becomes obvious what the maximum gain for a device is.

gustavecha, 2017-03-01:
Understood this clear explanation :)

I assumed RTL-SDR gain was limited to 40 by refering to max value used in SDR# / RTL_TCP with this output in console : "Setting gain value to index 40".
Are we sure it's only on RTL SDR API ? Spectrum analyzer works well with low gain

jopohl, 2017-03-01:
The "Setting gain value to 40" is message from URH. It has some maximum values for each device built in to prevent unneeded API calls and provide a maximum for the GUI elements.

You see only the frequency in spectrum analyzer: If you record with gain 40 you also see a signal, it is just quite weak. However, this is enough for FFT to perform its magic and show you the frequencies fo the signal.

gustavecha, 2017-03-01:
Regarding the 'Setting gain value to **index** 40', I was talking about what is shown in console when I use RTL_TCP with SDR#, it's wasn't about URH.
Such a message let me thought the gain was actually limited to 40 in RTL SDR API.

Ok for spectrum analyzer, seemed legit as I read it, Thx :)

jopohl, 2017-03-01:
Oh sorry I misread your first comment. Seems to be quite funny coincidence here:
- 40 was the value hardcoded for max gain for RTL-SDR in previous version
- "Setting gain value to index 40" indicates, that there are 40 gains available and RTL_TCP/SDR# choosing the last one in the array i.e. the greatest.

gustavecha, 2017-03-01:
Sounds good :)
And then, I definitely don't know what the highest value is, maybe 300 and accessed at the index 40

Have a nice evening