jopohl/urh #179: Unable to stop rad1o

ikarus23, 2017-03-02:
First of all: Thank you for this wonderful, well needed tool!

I'm using the [rad1o badge](https://rad1o.badge.events.ccc.de/), given out by the CCC at the Chaos Communication Camp 2016. This SDR is basically an HackRF clone. However, if I use the spectrum analyser of urh I'm unable to close the window after hitting the "stop receiving button", leaving urh in an unresponsive state. 
```
[WARNING] Receive buffer too small, skipping 31072 samples
[WARNING] Receive buffer too small, skipping 31072 samples
[WARNING] Receive buffer too small, skipping 31072 samples
[INFO] HackRF: Stopping RX Mode: Stopped receiving: Stop button clicked
[INFO] stopping HackRF rx mode (Stopped receiving: Stop button clicked)
[WARNING] closing because stop_rx_mode of HackRF is bugged and will not allow re receive without close
[INFO] HackRF: Attempting to close...
```
I know, the rad1o SDR is not officially supported, but maybe this behavior is well known to someone here and can be fixed. I would really like to use urh with my rad1o.

The rad1o does not run the latest big HackRF firmware update. Does urh only work with HackRFs that have been upgraded to the [2017.02.1 release](https://github.com/mossmann/hackrf/releases/tag/v2017.02.1)?

jopohl, 2017-03-03:
Thanks @ikarus23 , I am glad URH is helpful for you!

This is indeed an issue, that also affects the HackRF in rare cases (say 1 percent of the time). Problem is, that the HackRF ``` close ``` call sometimes blocks and never returns. Does this happen all the time for your rad1o badge or only sometimes?

In perspective, we should run the HackRF methods (open/close/start_rx etc.) in a seperate process to prevent the whole application from freezing.

ikarus23, 2017-03-03:
Unfortunately it happens all the time. Moving the methods to a separate process sound like a effective solution. Of course, some part of me will still be wondering what the real reason for this rare block behaviour is... ;)

9R, 2017-03-03:
I am using a rad1o as well and experienced similar problems while recording signals. In my case the recording dialog did not freeze 100% of the time but it seemed to be dependent  on the length of the recorded signal or file size. Recordings <20Mb never failed. I can a run couple more tests on Monday, when I am reunited with my rad1o.

vsboost, 2017-03-04:
Im getting this now on my hackrf when pressing stop.

[INFO] HackRF-start_rx_mode: Success
[INFO] HackRF: Stopping RX Mode: Stopped receiving: Stop button clicked
[INFO] stopping HackRF rx mode (Stopped receiving: Stop button clicked)
[WARNING] closing because stop_rx_mode of HackRF is bugged and will not allow re receive without close
[INFO] HackRF: Attempting to close...

Have to terminate urh

closed 2017-03-05 by jopohl

ikarus23, 2017-03-20:
Hi @jopohl, thanks for addressing this issue. Unfortunately the fix does not work for my rad1o. And even for the HackRF there is still a minor issue left.

1. Stopping the rad1o does not work. I'm not sure what is causing this. I will investigate this further, if I manage to find some spare time ;)
2. Stopping the HackRF works, however, not completely. In most cases the RX LED stays on, indicating that the HackRF is still receiving. Other tools like `osmocom_fft` do not show that behaviour. Is there no way to shutdown the HackRF like the `osmocom_fft` tool does?

Regarding 2.: This issue is common for some tools. It might be fixed in the future by the HackRF guys as a workaround. They are planning to implement an auto-reset, if the HackRF does not receive any USB signal for x seconds.

jopohl, 2017-03-20:
Hi @ikarus23 ,
I just plugged in my HackRF and could verify, that the RX LED stays on in most cases. I added a timeout in 225fbbc064bf5fc6ffd4fc8f445efc112a0c4ff2 in HackRF ``` stop_rx_mode() ``` and ``` close() ``` methods to address this issue. It works way better for me now. Maybe this also solves the problem with the unstoppable rad1o?

ikarus23, 2017-03-20:
I had the latest HackRF firmware. But your changes actually do the trick for the HackRF. I will check if they also fix the rad1o tomorrow.

jopohl, 2017-03-20:
That's cool! So the actual fix was to remove the timeout for ``` close() ```  in 2c8ce1e21f9aa6206e9d1a9436bf1cf148162b10. Hope this fixes rad1io as well!

ikarus23, 2017-03-21:
Hi @jopohl, now it works with the rad1o too! Thank you!