jopohl/urh #182: Losing data in Analysis tab

vsboost, 2017-03-05:
Eg. In the analysis tab i usually change to hex view so i see the data a little easier and i hide some rows that are not of interest, what i found though if i click a different tab say the generator tab and go back to the analysis tab all the data seems to vanish.

jopohl, 2017-03-05:
Can't reproduce this. Can you provide me with some more info?
1. How many rows does your protocol have?
2. How many files do you use?
3. What OS are you on?
4. Can you bring back your data if you right click to analysis table and choose "show all rows (reset n hidden)"?

vsboost, 2017-03-05:
1. 48 Rows
2. One file is loaded
3. Kali sp2 4.9.0-kali2-amd64 Debian 4.9.13-1kali1
4. If i select unhide x rows the data does not come back, i have to untick the file name in the box labeled "Protocols" and re tick it for the data to be displayed again.

So i have tried this on two separate files now, both of them exhibit the same symptoms.

-Load .cfile
-Switch to ASK
-Click Analysis tab
-Hide a few rows
-Click Generator tab
-Click back to Analysis tab
-Data gone

jopohl, 2017-03-05:
Can you tell me what your QT version is? If you use a recent version of URH you can find out with: ``` Help ``` -> ``` About Qt ```

vsboost, 2017-03-05:
5.7.1

jopohl, 2017-03-06:
Should work...I just made a possible fix in branch ``` gui_update_fix ```. Could you check out that branch and see if the problem persists?

vsboost, 2017-03-06:
Still the same, hope im doing it right, ill just go through the process with you.
git clone https://github.com/jopohl/urh.git
git checkout gui_update_fix

cd /urh/src/urh
./main.py

------------------------------------
Will not regenerate UI, because script cant be found. This is okay in release.
Using modules from /root/Downloads/urh/src
Could not find C++ extensions, trying to build them.
/tmp/hackrf_initkqy7ugtw.c:1:1: warning: return type defaults to ‘int’ [-Wimplicit-int]
 main (int argc, char **argv) {
 ^~~~
/tmp/hackrf_initkqy7ugtw.c: In function ‘main’:
/tmp/hackrf_initkqy7ugtw.c:2:5: warning: implicit declaration of function ‘hackrf_init’ [-Wimplicit-function-declaration]
     hackrf_init();
     ^~~~~~~~~~~

Found hackrf lib. Will compile with native hackrf support

/tmp/rtlsdr_set_tuner_bandwidthf109ksc3.c:1:1: warning: return type defaults to ‘int’ [-Wimplicit-int]
 main (int argc, char **argv) {
 ^~~~
/tmp/rtlsdr_set_tuner_bandwidthf109ksc3.c: In function ‘main’:
/tmp/rtlsdr_set_tuner_bandwidthf109ksc3.c:2:5: warning: implicit declaration of function ‘rtlsdr_set_tuner_bandwidth’ [-Wimplicit-function-declaration]
     rtlsdr_set_tuner_bandwidth();
     ^~~~~~~~~~~~~~~~~~~~~~~~~~

Found rtlsdr lib. Will compile with native rtlsdr support

running build_ext
Finalizing options
building 'urh.cythonext.path_creator' extension
creating build/temp.linux-x86_64-3.5
creating build/temp.linux-x86_64-3.5/src
creating build/temp.linux-x86_64-3.5/src/urh
creating build/temp.linux-x86_64-3.5/src/urh/cythonext
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-MLq5fN/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -c src/urh/cythonext/path_creator.cpp -o build/temp.linux-x86_64-3.5/src/urh/cythonext/path_creator.o -fopenmp
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
In file included from /usr/include/python3.5m/numpy/ndarraytypes.h:1788:0,
                 from /usr/include/python3.5m/numpy/ndarrayobject.h:18,
                 from /usr/include/python3.5m/numpy/arrayobject.h:4,
                 from src/urh/cythonext/path_creator.cpp:465:
/usr/include/python3.5m/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^~~~~~~
src/urh/cythonext/path_creator.cpp: In function ‘<built-in>’:
src/urh/cythonext/path_creator.cpp:2541:31: warning: ‘__pyx_v_tmp’ may be used uninitialized in this function [-Wmaybe-uninitialized]
                       #pragma omp for lastprivate(__pyx_v_chunk_end) firstprivate(__pyx_v_i) lastprivate(__pyx_v_i) lastprivate(__pyx_v_index) lastprivate(__pyx_v_j) lastprivate(__pyx_v_maximum) lastprivate(__pyx_v_minimum) lastprivate(__pyx_v_sample) lastprivate(__pyx_v_tmp) schedule(static)
                               ^~~
src/urh/cythonext/path_creator.cpp:2276:9: note: ‘__pyx_v_tmp’ was declared here
   float __pyx_v_tmp;
         ^~~~~~~~~~~
src/urh/cythonext/path_creator.cpp:2541:31: warning: ‘__pyx_v_sample’ may be used uninitialized in this function [-Wmaybe-uninitialized]
                       #pragma omp for lastprivate(__pyx_v_chunk_end) firstprivate(__pyx_v_i) lastprivate(__pyx_v_i) lastprivate(__pyx_v_index) lastprivate(__pyx_v_j) lastprivate(__pyx_v_maximum) lastprivate(__pyx_v_minimum) lastprivate(__pyx_v_sample) lastprivate(__pyx_v_tmp) schedule(static)
                               ^~~
src/urh/cythonext/path_creator.cpp:2273:9: note: ‘__pyx_v_sample’ was declared here
   float __pyx_v_sample;
         ^~~~~~~~~~~~~~
src/urh/cythonext/path_creator.cpp:2541:31: warning: ‘__pyx_v_minimum’ may be used uninitialized in this function [-Wmaybe-uninitialized]
                       #pragma omp for lastprivate(__pyx_v_chunk_end) firstprivate(__pyx_v_i) lastprivate(__pyx_v_i) lastprivate(__pyx_v_index) lastprivate(__pyx_v_j) lastprivate(__pyx_v_maximum) lastprivate(__pyx_v_minimum) lastprivate(__pyx_v_sample) lastprivate(__pyx_v_tmp) schedule(static)
                               ^~~
src/urh/cythonext/path_creator.cpp:2274:9: note: ‘__pyx_v_minimum’ was declared here
   float __pyx_v_minimum;
         ^~~~~~~~~~~~~~~
src/urh/cythonext/path_creator.cpp:2541:31: warning: ‘__pyx_v_maximum’ may be used uninitialized in this function [-Wmaybe-uninitialized]
                       #pragma omp for lastprivate(__pyx_v_chunk_end) firstprivate(__pyx_v_i) lastprivate(__pyx_v_i) lastprivate(__pyx_v_index) lastprivate(__pyx_v_j) lastprivate(__pyx_v_maximum) lastprivate(__pyx_v_minimum) lastprivate(__pyx_v_sample) lastprivate(__pyx_v_tmp) schedule(static)
                               ^~~
src/urh/cythonext/path_creator.cpp:2275:9: note: ‘__pyx_v_maximum’ was declared here
[... 61 lines cut ...]
/usr/include/python3.5m/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^~~~~~~
x86_64-linux-gnu-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -g -fdebug-prefix-map=/build/python3.5-MLq5fN/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security - [... cut ...]
copying build/lib.linux-x86_64-3.5/urh/cythonext/signalFunctions.cpython-35m-x86_64-linux-gnu.so -> src/urh/cythonext
copying build/lib.linux-x86_64-3.5/urh/cythonext/util.cpython-35m-x86_64-linux-gnu.so -> src/urh/cythonext
copying build/lib.linux-x86_64-3.5/urh/dev/native/lib/hackrf.cpython-35m-x86_64-linux-gnu.so -> src/urh/dev/native/lib
copying build/lib.linux-x86_64-3.5/urh/dev/native/lib/rtlsdr.cpython-35m-x86_64-linux-gnu.so -> src/urh/dev/native/lib
------------------------------------------
Above is the output from the first run

jopohl, 2017-03-06:
Yes you did it right way @vsboost . I am running out of ideas what could be the cause of this. I just installed latest Kali Linux to a VM and had no problems with GUI updates. Is your OS up to date?

vsboost, 2017-03-06:
Yea i update every few days, here is a link with video.
https://vid.me/yDqr

splotz90, 2017-03-06:
I can confirm this behavior. It occurs when the first row is hidden.

jopohl, 2017-03-06:
Thanks @splotz90 , hiding the first row does the trick!

commit 3bf6bb63 (2017-03-06): fix #182
closed 2017-03-06 by jopohl