jopohl/urh #184: Move to native C++ and drop Python wrappers

gustavecha, 2017-03-06:
Hi all,

Don't be affraid of such a strong issue title, I just wanted to get this discussion started about Python wrappers and performances.
It appears that using python wrappers can slow down any process using GNUradio components due to the overhead it adds.

Can this be possible, or even a good idea to move to native c++ and use Python for GUI only ?
It's not an easy thing and Python actually have important benefits to make URH cross platform, easy to deploy, etc that native c++ won't offer as this.

My knowledge about URH isn't exhautsive and I may miss some points about what really impact performances, feel free to close this if it is out of scope :)

All the best

jopohl, 2017-03-06:
Hi @gustavecha ,

the mission critical parts of URH (e.g. modulations / native device backends) are already implemented in C++. 

You can check this out e.g. in ``` src/urh/cythonext ``` or ``` src/urh/dev/native/lib ```.

andynoack, 2017-03-06:
Another point of view from a man with C programmers mindset:
I could have come to the same idea. Indeed, performance plays a big role for us when we implement things in URH. However, Python is not (any more?) that slow and we are doing all near realtime stuff in Cython anyway. I am sure we could improve the performance by 1-10% (probably not more), but then the community would have to wait another three years while we reimplement everything ;-) 
I think Python is the most economic choice at the moment.

jopohl, 2017-03-08:
I am closing this one. Feel free to open another issue, if you experience performance problems.

closed 2017-03-08 by jopohl

gustavecha, 2017-03-10:
Thank you guys for feedback

Indeed as @andynoack said, Python is the most economic choice for now and URH works just well with local devices.
My question was an open one to get a better understanding of how the project started

All the best