jopohl/urh #185: Save defaults

vsboost, 2017-03-06:
Hi there,

Any chance the program could save defaults for example i notice that FSK is default when opening a file, is it possible that if i select ASK that it will stay that way, and also for recording signals i think USRP is default, would love it if it remembers the last used device for me HackRF and the sample rate and frequency used.

jopohl, 2017-03-06:
Hey @vsboost ,

what you are describing is the project feature of URH. You can use it by opening a directory (File -> Open) and converting your folder to an URH project if asked. After this, your mentioned configurations will  be saved for this project.

vsboost, 2017-03-06:
Ahh thanks, didn't realize that.

closed 2017-03-06 by vsboost