jopohl/urh #190: Gnuradio interface is not working if both gnuradio and pothos are installed.

descite, 2017-03-08:
Hello. I have this problem in Windows10x64. Installed python 3.6 x64, installed Gnuradio (gnuradio_3.7.11_win64_avx2.msi) and urh. Checked that urh is enable to select airspy and sdrlay and record signal. 
But after installing PothosSDR (PothosSDR-2017.02.04-vc14-x64.exe) later, urh stops from working. It shows that gnuradio path is ok, but devices wont start with error:
`Traceback (most recent call last):
  File "C:\Program Files\Python36\lib\site-packages\urh\dev\gr\scripts\sdrplay_recv.py", line 23, in <module>
    from gnuradio import gr
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\gr\__init__.py", line 41, in <module>
    from runtime_swig import *
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\gr\runtime_swig.py", line 17, in <module>
    _runtime_swig = swig_import_helper()
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\gr\runtime_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_runtime_swig')
  File "C:\Program Files\GNURadio-3.7\gr-python27\lib\importlib\__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _runtime_swig
`

Seems that urh tries to use pothos instead of gnuradio?
After uninstalling PothosSDR, urh is working again. I tested multiple times to install/uninstall PothosSDR, each time with same result. If pothos is installed, urh gets this error, if not installed, urh is working.

andynoack, 2017-03-08:
Hi @descite 

that is indeed a problem. PothosSDR comes with an own bundled Gnuradio installation and changes all paths to its favor. Consequently some (maybe all) other programs based on Gnuradio do not work any more.

Our philosophy is to not interfere with any other installed program, i.e. we will not implement anything that prevents other installed programs from running (as PothosSDR does).

To solve your problem, you could try to set PYTHONPATH to its old destination. Maybe this is enough to run URH correctly, but no guarantee as Pothos might change even more system variables. Note that PothosSDR will probably not run any more when you do that.

jopohl, 2017-03-08:
Update @descite  and @andynoack : I found an easy way to fix this. Will push a fix soon, so Pothos can be installed side by side with URH on windows.

jopohl, 2017-03-08:
Just fixed this with https://github.com/jopohl/urh/commit/0e69fb9d0a42cf021c82dbf038e547a4496372a2.

closed 2017-03-08 by jopohl