jopohl/urh #198: Feature request - YARD Stick One

Lexus89, 2017-03-10:
Unfortunately there is no support for the YARD stick. It would be really great if support for this device is added!

vsboost, 2017-03-10:
+1 for yard stick one support, think it would be great especially for sending fuzzed messages.

jopohl, 2017-03-11:
I would love to implement support for YARD Stick One. Unfortunately, I do not own such as device and could not find any useful drivers via google. All I found was [rfcat](https://bitbucket.org/atlas0fd00m/rfcat) which appears to offer an interactive python shell. 

How you guys use your YARD Stick One? Does it work with an osmosdr block in GNU radio or you use  [rfcat](https://bitbucket.org/atlas0fd00m/rfcat) or a completely different tool?

Lexus89, 2017-03-11:
I use rfcat to send and receive signals

vsboost, 2017-03-11:
Hi
The YS1 itself is not an SDR more a radio modem where you send commands to it via rfcat, here is a link to some helper scripts that i use to give you an idea of how it works.
https://github.com/AndrewMohawk/RfCatHelpers

ikarus23, 2017-03-12:
100% agree with @vsboost. YS1 is not a SDR. It is just an "advanced breakout board" for the [TI CC1111](http://www.ti.com/lit/gpn/cc1110-cc1111) platform. It is by far not as flexible as a SDR. You have a limited set of input parameters you can define (like preamble, syncword, modulation, etc.) and it will decode the data for you.

I don't think there is a good way to integrate this into the workflow of URH. Well, at least the whole interpretation phase is pointless with the YS1. Only if you already know the parameters of the protocol you investigating, you can decode it with the YS1. However, using the fuzzer of URH for a YS1 might be handy. But on the other side, it is easy to implement a small fuzzer for YS1 using rfcat...

andynoack, 2017-03-15:
We are thinking about a YARDStick One support similar to the NetworkSDR device, i.e. without interpretation phase. However, we currently do not have YARDSticks at hands. Is someone connected to Great Scott Gadgets or can provide us with two sticks in another way?

vsboost, 2017-03-15:
I have reached out to Mike Ossmann the creator of the YS1, hopefully he might get in touch.

ikarus23, 2017-03-15:
I will meet him on Monday. I will ask him.

ikarus23, 2017-03-21:
Hi! I have great news that I wanted to share before I go to bed ;) Michael Ossmann (@mossmann) just handed me over two YARD Stick One for you! will contact you via email regarding the shipping.

Thanks to @mossmann for helping out the URH developers in improving this great software!

vsboost, 2017-03-22:
Great stuff, well done.

Lexus89, 2017-03-22:
Wow really awesome! Thanks @ikarus23 and @mossmann !

andynoack, 2017-04-20:
We have experiemented with the YARD Sticks for a while (thanks again!) and came to the conclusion that a YARD Stick support does not comply with the URH philosophy. Why that?
- YARD Stick is designed to work with already known protocols (byte patterns), URH instead helps to find out the byte patterns and protocol fields
- YARD Stick does not output I/Q data, so we cannot handle it as a SDR (like all other devices we are supporting)
- RfCat (or Rflib) is used as interface to the YARD Stick. However, RfCat (hopefully not the hardware) does not run stable when receiving longer byte streams. There are USB timeouts that prevent from using the byte stream in a reliable way.

Never say never, maybe we support the YARD Stick in some way in the future. To everyone who is using the YARD Stick to pentest/reverse engineer protocols. Which kind of support would help you? A plugin showing the approximated YARD Stick (RfCat) parameters when sniffing with e.g. a RTLSDR? Please be precise and give some good arguments :)

Best @jopohl and myself

vsboost, 2017-04-20:
So this is how i would love to see it work.
1) Sniff / Record signal using rtl / hackrf / sdrplay / usrp / airspy
2) Analyse signal, get your modulation format / 1s and 0s / get your baud/symbol rate etc etc
3) Fuzz or change data to suit.
4) Send via Rfcat device / YS1.

I really don't care for the RX side using Rfcat/YS1, in my opinion you need to use SDR to know the signal first.

Hope this makes sense.

andynoack, 2017-04-20:
Ok, we have an idea...Stay tuned!

Lexus89, 2017-04-20:
+1 Exactly what is said above. 

I have an RTL-SDR for grabbing the signal and it would be nice if you could replay or fuzz using this signal directly with the yard stick. Managing it all from a single tool, URH, makes things so smooth.

andynoack, 2017-04-22:
So guys, I just commited a first shot of a RfCat plugin, ready for alpha testing! In the generator tab there will be a button that allows to send what is currently displayed via RfCat (e.g. with YS1). Does this work for you?

vsboost, 2017-04-22:
Sweet this works for me... does the button do anything at the moment? is there anything i can test on my end?

andynoack, 2017-04-22:
The button should start a rfcat process in background (URH should be able to launch 'rfcat -r' from current path) and communicate with it. If you drag some messages in the generator field and push the button, it should send the according encoded information out via the YS1. While doing that, you should see according rfcat commands in the debug/console output.
I was able to turn on my wireless room lights with it. Can you verify that this works for you? Maybe (if it transmits data) check if the sent data is encoded and modulated in the right way using e.g. a rtlsdr?

vsboost, 2017-04-23:
MMM, when i tried i couldn't see anything happening, have to try again and let you know.

vsboost, 2017-04-23:
Still cant see anything happening when i press the send button the console shows no messages and YS1 does not TX, rfcat is in path and working.

vsboost, 2017-04-23:
Is there anything else i need to do to the data in the generate tab prior to sending?

andynoack, 2017-04-23:
I just added some debug messages to localize the problem you have. Can you try with the latest code?

vsboost, 2017-04-23:
In the console all i get is

Using modules from /root/Downloads/urh/src
[DEBUG] Successfully opened RfCat (rfcat)

When i press the send via rfcat button, but still no TX

andynoack, 2017-04-23:
Can you verify the rfcat process running (ps ax | grep rfcat)? What kind of data do you have in your generator window?

vsboost, 2017-04-23:
12328 pts/0    Sl+    0:00 /usr/bin/python /usr/local/bin/rfcat -r
12491 pts/1    S+     0:00 grep rfcat

Just sending some 10101010101010101 nothing fancy

andynoack, 2017-04-23:
Ok...try the latest code...it is very verbose now

vsboost, 2017-04-23:
Thanks for the extra debug...

Using modules from /root/Downloads/urh/src
[DEBUG] Successfully opened RfCat (rfcat)
[DEBUG] Modulation = MOD_ASK_OOK
[DEBUG] Frequency = 433920000.0
[DEBUG] Sample rate = 1000000
[DEBUG] Bit length = 532
MSG: 1001100110011001101001100101010110101001100110010101100110100110100101101010100110101001010101010101101010100101101010100101010110010101101010100101010101010101

However still no TX from the yard stick one, im going to try on another system.

vsboost, 2017-04-23:
Tried on a different system same deal, no TX.
No issues with my YS1 as i can tx with it using rfcat directly so not sure what else i can do.

andynoack, 2017-04-23:
It is very hard to localize the problem. I have another idea...latest code update is online

vsboost, 2017-04-23:
We have TX :-)

Thanks very much.

andynoack, 2017-04-23:
Well ok, identified the problem! This is very weird and I think it is related to rfcat. I will cleanup the code in the next days and prepare a release.

vsboost, 2017-04-23:
Sweet, now to have a play.

Thanks again.

vsboost, 2017-04-23:
Great work Andy

Lexus89, 2017-04-23:
I just successfully replayed a message with YS1, great work thanks!

andynoack, 2017-04-26:
I close the issue because we released the rfcat-plugin now, see https://github.com/jopohl/urh/releases/tag/v1.6.2.5

closed 2017-04-26 by andynoack