jopohl/urh #199: RTL-TCP ran out of input

gustavecha, 2017-03-10:
Hi,

While I was trying to test RTL-TCP reception with my freshly upgraded URH, this error crash Python when clicking the "start" button of spectrum analyzer : 

> [INFO] Initialized receiving buffer with size 0.76MB
> [INFO] RTLSDRTCP: Starting RX Mode
> Traceback (most recent call last):
>   File "c:\program files\python\python35\lib\site-packages\urh\controller\SpectrumDialogController.py", line 77, in on_start_clicked
>     self.device.start()
>   File "c:\program files\python\python35\lib\site-packages\urh\dev\VirtualDevice.py", line 360, in start
>     self.__dev.start_rx_mode()
>   File "c:\program files\python\python35\lib\site-packages\urh\dev\native\Device.py", line 247, in start_rx_mode
>     self.receive_process.start()
>   File "c:\program files\python\python35\lib\multiprocessing\process.py", line 105, in start
>     self._popen = self._Popen(self)
>   File "c:\program files\python\python35\lib\multiprocessing\context.py", line 212, in _Popen
>     return _default_context.get_context().Process._Popen(process_obj)
>   File "c:\program files\python\python35\lib\multiprocessing\context.py", line 313, in _Popen
>     return Popen(process_obj)
>   File "c:\program files\python\python35\lib\multiprocessing\popen_spawn_win32.py", line 66, in __init__
>     reduction.dump(process_obj, to_child)
>   File "c:\program files\python\python35\lib\multiprocessing\reduction.py", line 59, in dump
>     ForkingPickler(file, protocol).dump(obj)
> TypeError: can't pickle _thread.lock objects
> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
>   File "c:\program files\python\python35\lib\multiprocessing\spawn.py", line 106, in spawn_main
>     exitcode = _main(fd)
>   File "c:\program files\python\python35\lib\multiprocessing\spawn.py", line 116, in _main
>     self = pickle.load(from_parent)
> EOFError: Ran out of input

As you can understand, I run URH on windows 10 64 bits

I type theses parameters in the spectrum window :
- Device = RTL-TCP
- IP adress = <remote ip address>
- Port number = Correspponding listening port
- Frequency = 433,920M
- Sample rate = 1,000M
- Bandwidth = 1,000M
- Gain = 20

Am I doing anything wrong ?

Anyway thank you for any hint provided :)

andynoack, 2017-03-11:
That looks like a multiprocessing problem on the first sight. However, would you try to set the 'ip address' field to either 127.0.0.1 or the IP address of your local network interface (assuming that rtl_tcp runs on the same computer). Furthermore the standard TCP port of rtl_tcp is 1234, try that one too.

gustavecha, 2017-03-11:
Hi @andynoack 

The rtl_tcp server runs on a different device in my attic, then the ip address have to be different than the localloop or the local interface of the machine where urh runs.
The port was indeed 1234.

Do you need more input to understand what the problem is or is this a configuration mistake on my side ?

andynoack, 2017-03-11:
I think, you were right. We will have a look on it!

andynoack, 2017-03-11:
Hey @gustavecha 
can you try the 'enhance_device_config' branch of URH and check whether the error is fixed there?

jopohl, 2017-03-12:
This seems to be a windows exclusive error. It appears on both ``` master ``` and ``` enhance_device_config ``` branch, so no need to test that out anymore @gustavecha . I get the same error as you. We will have a look at this.

closed 2017-03-12 by jopohl

gustavecha, 2017-03-12:
Hi guy,
Tested 1.6.1 this evening

RTL-TCP receive signal normally on windows now,
how fast you are :)

PR #187 (2017-03-12): enhance device config
- [x] Save Gains when dialog closed
- [x] remove _max_gain / _max_sample rate etc. from native device class
- [x] pass HackRF gain parameters to GR backend
- [x] hide HackRF baseband gain for Send
- [x] update HackRF IF Gain depending on send/recv
- [x] Fix high GUI gain for RTLSDR -> calc *10 for API
- [x] Add tooltips for Gain / IF Gain / Baseband Gain
- [x] use dict for saving paramaters in project manager
  - [x] save currently saved parameters
  - [x] save freq correction
- [x] fix #133 (support for USRP B200/B210)
- [x] fix #136 (HackRF Gain sliders & LNA button)
-  [... cut]