jopohl/urh #200: Crashes by using the Decoding-Tool

ikarus23, 2017-03-10:
Open URH --> Edit --> Decoding... --> Drag & Drop External Program into Decoder --> Drag & Drop External Program back into Base Functions --> Crash.
```
Will not regenerate UI, because script cant be found. This is okay in release.
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/urh/controller/DecoderWidgetController.py", line 397, in on_decoder_chain_current_row_changed
    self.set_information(2)
  File "/usr/lib/python3.6/site-packages/urh/controller/DecoderWidgetController.py", line 415, in set_information
    element = self.ui.decoderchain.currentItem().text()
AttributeError: 'NoneType' object has no attribute 'text'
[1]    6671 abort (core dumped)  urh
```

Open URH --> Edit --> Decoding... --> Drag & Drop Substitution into Decoder --> Fill out some rows of the Substitution block --> Save As --> Enter Name --> Hit OK --> Crash
```
Will not regenerate UI, because script cant be found. This is okay in release.
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/urh/controller/DecoderWidgetController.py", line 397, in on_decoder_chain_current_row_changed
    self.set_information(2)
  File "/usr/lib/python3.6/site-packages/urh/controller/DecoderWidgetController.py", line 415, in set_information
    element = self.ui.decoderchain.currentItem().text()
AttributeError: 'NoneType' object has no attribute 'text'
[1]    6774 abort (core dumped)  urh
```

And I don't remember what I did there, but I got this this crash too:
```
Will not regenerate UI, because script cant be found. This is okay in release.
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/urh/controller/DecoderWidgetController.py", line 766, in dropEvent
    self.chainoptions.pop(self.ui.decoderchain.item(self.ui.decoderchain.active_element).text(), None)
AttributeError: 'NoneType' object has no attribute 'text'
[1]    6551 abort (core dumped)  urh
```

I hope this help on the way to improve this great tool!

andynoack, 2017-03-11:
Thanks for the advice and the very good description of the process leading to the error. That should help us finding the problem!

jopohl, 2017-03-11:
Thanks again for this very detailed report @ikarus23 . Helped me alot fixing this bug.

commit dc466245 (2017-03-11): fix #200
closed 2017-03-11 by jopohl