jopohl/urh #202: RTL-TCP gains adjustments

gustavecha, 2017-03-12:
Hi,

Tested RTL-TCP successfully this evening with URH 1.6.1.
Here are my comments about gain values for this particular device.
RTL-SDR v3 dongle is used on my rtl-tcp server.

Here is a rtl-tcp output with SDR# client connected:
> Use the device argument 'rtl_tcp=x.x.x.x:1234' in OsmoSDR (gr-osmosdr) source
> to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
> client accepted!
> set sample rate 900001
> Exact sample rate is: 900001.019241 Hz
> set freq correction 0
> set freq 123456789
> set agc mode 0
> set gain mode 1
> set tuner gain by index 17
> 

And here is a rtl-tcp output with URH client connected with gain value 40 in signal recorder :
> Use the device argument 'rtl_tcp=x.x.x.x:1234' in OsmoSDR (gr-osmosdr) source
> to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
> client accepted!
> set freq 123456789
> set sample rate 900001
> Exact sample rate is: 900001.019241 Hz
> set gain 50
> set freq correction 0
> set direct sampling 0
> Disabled direct sampling mode

Gain mode and values between two clients aren't the same
SDR# seems to use predefined gain values by giving their indexes.

Spectrum analyzer gives pretty accurate output when high gain values are selected (x10 in the rtl-tcp output). Using 34 for gain instead of 32 gives too much significant changes on spectrum. This have to be more progressive (< 32 : noise level only, > 34 signals are seen really strong above noise level)

I listen to frequencies where trunk messages are received every 30 seconds and signal recorder dosen't catch them at all while SDR# shows me those pulses by 25dB above noise level.

Furthermore signal recorder doesn't apply the x10 factor to gain values like spectrum analyzer does (it may be the reason).

Anyway thank you for improvements and support for rtl-tcp last week. Things go impressively fast on this project :)

andynoack, 2017-03-12:
Hi @gustavecha 
I just wrote a simple fix for your problem, should be fixed now. You are welcome!

PS (gain control): we are not using the predefined gain values for rtlsdr as we like to have a highly generic control dialog that supports a bunch of various SDRs (all of them have some specialities) in a similar way.

jopohl, 2017-03-13:
Hi @gustavecha 

problem with RTL-SDR is, that different sticks support different gains. There is an API method to get the gains available on a stick, which produces this for my device:
```
[0, 9, 14, 27, 37, 77, 87, 125, 144, 157, 166, 197, 207, 229, 254, 280, 297, 328, 338, 364, 372, 386, 402, 421, 434, 439, 445, 480, 496]
```
Note, that all gain values are calculated *10. This is the way, the RTL-SDR API demands and outputs them.

Here you can also see the reason for your described jumpy behaviour: If I put the gain in my example e.g. from 30 (mapped to 27) to 32 (mapped to 37 by API) I experience a much larger jump, than I  would expect. The certain limits may vary from device to device.

A solution for this is to ask the user for an index in above gain array and not the exact gain value. This is the way, that SDR# goes. However, as @andynoack pointed out, this would conflict with the usability for other SDRs in URH. 

I am open for suggestions on this.

jopohl, 2017-03-13:
I just published version 1.6.1.1 which fixes this issue.

closed 2017-03-13 by jopohl

gustavecha, 2017-03-13:
Hi :)

Understood the need to have generic gain controls to gat many devices up & running in urh.
If you have a class or even .py script for each particular device, can the use of predefined gains be included into ?
I mean : gain scroller value can be abstracted and select an index instead of a value and direct input numeric stepper can be locked when a given device doesn't accept free gain values.
It will improve devices integration and prevent user from typing wrong gains or issue signal strengh problems.
I think this abstraction level is highly recommended just to manage min and max values which can differ between RTL-SDR sticks and USRPs.

Request for possible gain values against the device API when connecting would be icing on the cake

Anyway, I'm away from my lab this week and can't wait to test this fix by friday :)