jopohl/urh #203: Feature request "Signal Statistics"

vsboost, 2017-03-12:
Hi there,

Was just thinking to myself that it would be cool to have a tab dedicated to the statistics of the data for example say i had 100 rows of data and say i am viewing it in ASCII or HEX that possibly a "Signal Statistics" tab could tell me how many times a signal repeats, show me a list of all say ASCII or HEX chars used and how many times they are seen in the signal.

Like

A -- Seen 2033 times
0 -- Seen 200 times
7 -- Seen 123 times
1 -- Seen 3 time
F -- Seen 91 times

Row 30 - 20 - 19 -- Identical

Just a thought anyway.

Thanks again, this is now my go to program for signal hacking.

andynoack, 2017-03-12:
Hey @vsboost
nice to hear that! Do you have a special use case for this? I mean, the implementation would not be a big problem (worst thing is probably the GUI integration). Till now we haven't missed such a feature when analyzing signals. Could you point out the demand/motivation for that, maybe some details for us to better understand?

vsboost, 2017-03-13:
I think it would help in creating more efficient fuzzed messages knowing that only certain values are used at any particular bit position.

So for example if i were to fuzz position 3&4 of a particular transmission rather then sending every possible hex value it would be nice to know based on the statistics of the recorded sample that the only values used are say A,B,1,2,3

Hope that makes sense.

andynoack, 2017-03-13:
Hi @vsboost 
we discussed it and will implement it in a future release!

vsboost, 2017-03-13:
Sounds good, thanks guys :-)