jopohl/urh #206: Drop pyqt5 requirement from setup.py?

yarda, 2017-03-14:
Hi,

I packaged urh for fedora [1], but I encountered the following error which prevents urh from running:
$ urh
Traceback (most recent call last):
  File "/usr/bin/urh", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2958, in <module>
    @_call_aside
  File "/usr/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2944, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2971, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3.5/site-packages/pkg_resources/__init__.py", line 635, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3.5/site-packages/pkg_resources/__init__.py", line 943, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3.5/site-packages/pkg_resources/__init__.py", line 829, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'pyqt5' distribution was not found and is required by urh

I am not python expert, but it seems that pyqt5 doesn't have the egg-info (at least on Fedora and it also seems that probably also not on Gentoo). That's why the 'pyqt5' distribution cannot be found. Well, this could be distro's bug, but I am currently not sure. By just removing the following from the setup.py:
-try:
-    import PyQt5
-except ImportError:
-    install_requires.append("pyqt5")

it starts working. As we are packaging it through the distro's package management system (RPM in case of Fedora) the requirements of pyqt5 are already correctly handled by RPM.

[1] https://bugzilla.redhat.com/show_bug.cgi?id=1432076

jopohl, 2017-03-14:
Hi @yarda ,

thanks for your work to create a fedora package of URH!

The rationale behind the four lines you mentioned is to add pyqt5 as a dependency for pip if it is not already installed on the system.

The error you posted indicates, that PyQt5 was not present during installation of URH. Is it possible to enforce PyQt5 installation __before__  URH gets installed? This way, the ImportError should not be triggered and pyqt5 will not be added to ``` install_requires ```.

yarda, 2017-03-14:
Ah, thanks, nice workaround. I added it to the build root and it works as expected.

closed 2017-03-14 by yarda