jopohl/urh #207: Feature Request

JamesonBeebe, 2017-03-16:
I love Hacking like a boss with this app! I think what would make this more boss-like is to have several more different modulation options like "PCM" among others in the Interpretation tab.

andynoack, 2017-03-17:
Thank you for bringing up this issue! Can you provide us with some raw signals using other modulations than the provided ones?

(Currently only modulations with modulation index 2 are supported, as other modulations like QAM will require some serious modifications in the core code and some very good ideas to save the usability.)

Supergoldfish, 2017-03-21:
Demodulating PIE encoding would also be very useful.  for example 1 bits are 400uS long and 0 bits are 200us long.  Sent at a baud rate of 600uS leaving gaps between bits. 
[image]

Demodulation would be something like checking the pulse width every 600uS.  Could have a field for 1 bit width and 0 bit width. 

PS, this package is really getting useful!  Thank you.

andynoack, 2017-03-21:
Hey @Supergoldfish ,
you can already do that by choosing a smaller bitlength (e.g. 200us) and defining a custom decoding (e.g. remove carrier or substitution). Does that help you?

JamesonBeebe, 2017-03-21:
This is a signal I'm trying to demodulate. However I can't quite tell the modulation type, but I'm thinking it's some sort of PFM. Also the device broadcasting is cheap remote control unit, so the signal isn't the best quality. Many remotes can be used with this device, so I'm pretty sure it's just blasting out a repeated signal until the receiver finally acknowledges it.

Any suggestions on what type of signal it is, and how we can demodulate it?

[New HackRF 433_806MHz 2MSps 2MHz_complex_Sample01_Test.zip](https://github.com/jopohl/urh/files/858669/New.HackRF.433_806MHz.2MSps.2MHz_complex_Sample01_Test.zip)

andynoack, 2017-03-21:
It rather looks like an ASK for me, but you may also be right with a PSK assumption. However, it seems a bit overmodulated to me, have you tried to reduce the gain or increase the distance to the device?

Supergoldfish, 2017-03-22:
Thanks for the tip!  Substitution did the job.

jopohl, 2017-03-22:
@JamesonBeebe additional to lowering the gain as @andynoack mentioned you may also try tuning to another center frequency. 433.92 MHz may be a good start.

andynoack, 2017-04-20:
I think we can close here. Just comment if you need further help on this topic and we will reopen.

closed 2017-04-20 by andynoack

aedrax, 2019-02-21:
Is there a sneaky way to do 8PSK or QAM?

jopohl, 2019-02-22:
Not right now, but we are working on a support.