jopohl/urh #209: Noisy spectrum analyzer start

gustavecha, 2017-03-18:
Hi,

I going through weekly test of URH 1.6.1.8 :)
In spectrum analyzer, with RTL-TCP device.

Once start button pressed, pretty high noise level quickly appears and then disappear. It cause a bad memory effect on the red part of the FFT.
Peak detection and visualisation afterwards is hard.
I think it's about the parameters sent to rtl-tcp (gain for instance) a bit late which cause this artefact at the beginning.

Is this possible to skip the x first sample sent by device ?
Or offer to user the possibility to clean the FFT memory while listening ?
Currently it's only possible to do so when spectrum analyzer is not listening.

jopohl, 2017-03-19:
Hi @gustavecha ,
thanks for using your precious weekend time for testing URH. :)

What you describe is indeed an issue. I favor the idea of giving the user the possibility to clear the FFT cache while listening. I will take a look at that.

closed 2017-03-19 by jopohl

gustavecha, 2017-03-21:
Hi

Why not for spectrum analyser, but the problem is also seen on signal recording with a big pulse at the very begining.
This make me say the root cause is elsewhere.

It really sounds like that rtl-tcp sends packet prior to get all parameters and such samples should be ignored (not sure of frequency, and gain)

Rtl-tcp doesn't have a start method to be called once all settings have been sent ?

andynoack, 2017-03-21:
rtl_tcp is quite simple, it sends samples after you have connected to the process. No matter, whether your configurations are completed or not. However, when wiretapping the communication with wireshark you can see that the sample transmission normally begins after all settings were done (due to a very fast configuration of all settings).
I think, this overmodulation has something to do with the hardware itself, since also Ettus USRP N210s or HackRFs (i.e. most of the SDRs we have worked with) show a similar behaviour.

commit d45bdf37 (2017-03-19): fix #209