jopohl/urh #210: Spectrum analyzer needs a lot of CPU (with rtl-tcp)

gustavecha, 2017-03-18:
Hi,

Currently using the 1.6.1.8 version of URH

Starting spectrum analyzer with RTL-TCP + 900 kSps consume a lot of CPU.
My 8-core Intel i7 3820QM 2.7 GHz gets 7 cores flooded à 100%

Consecutively, I see a low pretty refresh rate : a pulse sent every 30s isn't seen after 2 minutes of waiting

I know signal processing needs a lot of resources, but is this case normal ?

jopohl, 2017-03-19:
Hi @gustavecha ,

I was not able to reproduce this problem. However, a 2 minute delay is not acceptable! I made the ``` SPECTRUM_BUFFER_SIZE ``` configurable in ``` constants.py ``` with a11838fc4fcfb7dbf483e04ad9b6e2d8ff073f03. It is currently set to 10000. Could you set it to a lower value and see, if this solves the problem? If so, we can make this configurable in Options GUI.

k9wkj, 2017-03-20:
is this just rtl_tcp rearing its horrible ugly head?
ive only gotten rtl_tcp to run well on a Pi on x86 it was a stuttery
horrible mess
and of course over wifi its unuseable

gustavecha, 2017-03-21:
@jopohl, what was the configuration used to test?
Linux, windows? What cpu?

On my machine (windows 10 x64 i7 3820QM), urh is seen at 100% cpu while listening rtl-tcp with no real-rime and sdr# only needs 30% with full realtime
I'll test lowering buffer size by the end of this week to see how it changes

jopohl, 2017-03-21:
@gustavecha I tested on both, on Linux and on a Windows VM with a Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz .
I also tested on my older laptop (linux only here) with Intel(R) Core(TM) i5 CPU  M 520  @ 2.40GHz.

In all cases, Spectrum Analyzer worked like a charm.

closed 2017-04-12 by andynoack

gustavecha, 2017-05-01:
Hi,

Installed 1.6.2.6 version today and no delays are observed with this version, instead of before with 1.6.2.0
Optimization sounds good guys :)

7 cores out of 8 are actually used, it may be interesting but not necessarily to make the spectrum drawing a bit lighter

All the best