jopohl/urh #212: Shall we implement a Windows/OSX installer? (Thumbs up for yes)

andynoack, 2017-03-20:
The question is whether the current installation routine is too complicated on Windows and OSX. If you really need a installer or think, that might be useful, leave a comment or thumbs up.

vsboost, 2017-03-20:
Windows installer would be nice for those who find it difficult with building/compiling software.

andynoack, 2017-03-20:
Do you think there is a large interception of people that find it difficult building software and use SDRs?

gustavecha, 2017-03-20:
Hi,

Windows installation is really easy :
> pip install urh

Do we really need an installer for that ?
I guess no

vsboost, 2017-03-20:
Im a member of a few SDR groups and i always hear them complain about linux software or find it difficult to build for windows and install 50 different things just to get it going.

vsboost, 2017-03-20:
Not that it bothers me, i have dual boot windows/linux.

ikarus23, 2017-03-22:
I don't think a Windows installer is needed. It is likely that users interested in the features that URH provide (blind signal analysis, fuzzing, etc.) already use Linux.

andynoack, 2017-03-27:
Thanks for the feedback! I will close here.

closed 2017-03-27 by andynoack