jopohl/urh #213: Splitting Messages

ikarus23, 2017-03-22:
Hi, I came across a device sending a long sequence of packets. URH puts the sequence into one row of the analysis tab. What I really want to be able to do is to split a row at any given bit into two rows. Did I miss something? I this already possible?

Another nice thing would be to automate this splitting process by providing a sequence of bits that indicate the start of a new packet (row).

jopohl, 2017-03-22:
Hi  @ikarus23 ,

this is already possible with the Message Break Plugin. You can activate it via Options -> Plugins. After activation you have a new action in the context menu of analysis table.

Automating this is indeed a really nice idea. We will have a look at this!

ikarus23, 2017-03-22:
Oh, I totally missed out on the plugins. Thank you for pointing that out.

Just out of curiosity: Is there a reason why this is a plugin and not a default feature of URH?

jopohl, 2017-03-23:
We always try to find a balance between useability and features. We want to make it as easy as possible for a new user to get started with URH. Therefore, we put some functions into plugins, to prevent having context menu beasts.

In the end, I think the MessageBreak Plugin tends to be more a pro feature, as you do not need it for most protocols.