jopohl/urh #218: Name suggestions on saving files

ikarus23, 2017-03-30:
Hi, when I record a sample it is suggesting to name my file `HackRF 433_923MHz 4MSps 4MHz_complex`. The suggested extension (filter) in the saving dialog is `*.complex`. However, my files do not get the extension `.complex`. Maybe this is because the suggested name is `_complex`?

jopohl, 2017-03-30:
Hi @ikarus23 

I can not reproduce this behaviour. Can you tell me which OS and Qt version you use?

ikarus23, 2017-03-30:
I'm using Arch Linux with Qt version 5.8.0. I will try if I get the same behavior when I'm building URH straight from the github master.

jopohl, 2017-03-30:
Well that is really strange, because I use exactly the same setup.

Can this be related to some desktop environment "features"? What desktop environment to you use? XFCE here.

ikarus23, 2017-03-31:
I'm not using any desktop environment. Just die Awesome window manager.

I checked and I got the same behavior when I'm running URH from git.

Maybe my bug report was just shitty and we have a misunderstanding here :)

1. Run URH
2. File -> Record siganl
3. Press Record
4. Press Stop
5. Press Save
6. Press Save in file destination dialog

Expected behavior: The file should be saved as `HackRF 433_923MHz 4MSps 4MHz.complex`
Actual behavior: The file is saved as `HackRF 433_923MHz 4MSps 4MHz_complex`
Because of the `_complex` the `*.complex` filter will hide the saved files upon opening a sample.

jopohl, 2017-03-31:
Thanks for clarifying this @ikarus23 . Although I was still not able to reproduce the behaviour, I wrote a fix that should fix this problem. Can you verify this?

ikarus23, 2017-03-31:
Thanks @jopohl, it works now!

closed 2017-03-31 by ikarus23