jopohl/urh #219: How to upgrade URH in Windows

ngmacha, 2017-03-31:
Maybe the answer to this question is obvious, but how do I upgrade in Windows to the latest build? Currently I'm running version 1.4.7, and looks like there are some improvements since then - but I don't want to mess up the running configuration... 
Thank you in advance

jopohl, 2017-03-31:
Hi @ngmacha 

you can upgrade via pip with
``` pip3 install --upgrade urh ``` 

I will add a hint for this to the readme.

ngmacha, 2017-03-31:
Hi @jopohl 
Thank you very much for a quick response, I was able to upgrade to URH ver 1.6.1.9 now! However `pip3 install --upgrade urh` didn't work for me (Fatal error in launcher: Unable to create process using ""), but `python3 -m pip install --upgrade urh` did the job.

I think maybe best to add the "How to upgrade..." line to Help menu?

And thank you for a very interesting tool! 

Mac Ha

jopohl, 2017-03-31:
Thanks @ngmacha . I added your line to the readme.

closed 2017-03-31 by jopohl