jopohl/urh #220: Errors running URH on Armbian for Pine64

ngmacha, 2017-03-31:
I'm trying to install and run URH on the Pine64 board under Armbian (https://www.armbian.com/pine64/), which is a port of Ubintu 16.04.2 LTS. I have a Pine64 board with the touch screen, Armbian running quite fine, so with URH running it could be a good handheld radio hacking tool. However trying to run directly from source gives this error:

> nmha@pine64:~/urh/src/urh$ ./main.py
Will not regenerate UI, because script cant be found. This is okay in release.
Using modules from /home/nmha/urh/src
Could not find C++ extensions, trying to build them.
/tmp/rtlsdr_set_tuner_bandwidthkkc2640w.c:1:1: warning: return type defaults to ‘int’ [-Wimplicit-int]
 main (int argc, char **argv) {
 ^
/tmp/rtlsdr_set_tuner_bandwidthkkc2640w.c: In function ‘main’:
/tmp/rtlsdr_set_tuner_bandwidthkkc2640w.c:2:5: warning: implicit declaration of function ‘rtlsdr_set_tuner_bandwidth’ [-Wimplicit-function-declaration]
     rtlsdr_set_tuner_bandwidth();
     ^
/usr/bin/ld: cannot find -lrtlsdr
collect2: error: ld returned 1 exit status
Trying fallback for rtlsdr
/tmp/rtlsdr_get_device_namewxjdoj9n.c:1:1: warning: return type defaults to ‘in ’ [-Wimplicit-int]
 main (int argc, char **argv) {
 ^
/tmp/rtlsdr_get_device_namewxjdoj9n.c: In function ‘main’:
/tmp/rtlsdr_get_device_namewxjdoj9n.c:2:5: warning: implicit declaration of function ‘rtlsdr_get_device_name’ [-Wimplicit-function-declaration]
     rtlsdr_get_device_name();
     ^
/usr/bin/ld: cannot find -lrtlsdr
collect2: error: ld returned 1 exit status
/tmp/hackrf_init36jcflht.c:1:1: warning: return type defaults to ‘int’ [-Wimplicit-int]
 main (int argc, char **argv) {
 ^
/tmp/hackrf_init36jcflht.c: In function ‘main’:
/tmp/hackrf_init36jcflht.c:2:5: warning: implicit declaration of function ‘hackrf_init’ [-Wimplicit-function-declaration]
     hackrf_init();
     ^
/usr/bin/ld: cannot find -lhackrf
collect2: error: ld returned 1 exit status
Skipping native support for hackrf
running build_ext
Finalizing options
Traceback (most recent call last):
  File "setup.py", line 142, in <module>
    'urh = urh.main:main',
  File "/usr/lib/python3.5/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.5/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.5/distutils/dist.py", line 973, in run_command
    cmd_obj.ensure_finalized()
  File "/usr/lib/python3.5/distutils/cmd.py", line 107, in ensure_finalized
    self.finalize_options()
  File "setup.py", line 52, in finalize_options
    import numpy
  File "/usr/lib/python2.7/dist-packages/numpy/__init__.py", line 180, in <module>
    from . import add_newdocs
  File "/usr/lib/python2.7/dist-packages/numpy/add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "/usr/lib/python2.7/dist-packages/numpy/lib/__init__.py", line 8, in <module>
    from .type_check import *
  File "/usr/lib/python2.7/dist-packages/numpy/lib/type_check.py", line 11, in <module>
    import numpy.core.numeric as _nx
  File "/usr/lib/python2.7/dist-packages/numpy/core/__init__.py", line 14, in <module>
    from . import multiarray
ImportError: cannot import name 'multiarray'
Traceback (most recent call last):
  File "./main.py", line 123, in <module>
    main()
  File "./main.py", line 74, in main
    from urh.controller.MainController import MainController
  File "/home/nmha/urh/src/urh/controller/MainController.py", line 11, in <module>
    from urh.controller.CompareFrameController import CompareFrameController
  File "/home/nmha/urh/src/urh/controller/CompareFrameController.py", line 5, in <module>
    import numpy
  File "/usr/lib/python2.7/dist-packages/numpy/__init__.py", line 180, in <module>
    from . import add_newdocs
  File "/usr/lib/python2.7/dist-packages/numpy/add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "/usr/lib/python2.7/dist-packages/numpy/lib/__init__.py", line 8, in <module>
    from .type_check import *
  File "/usr/lib/python2.7/dist-packages/numpy/lib/type_check.py", line 11, in <module>
    import numpy.core.numeric as _nx
  File "/usr/lib/python2.7/dist-packages/numpy/core/__init__.py", line 14, in <module>
    from . import multiarray
ImportError: cannot import name 'multiarray'

I tried to install URS using `pip3 install urh`, without errors, but running it still failed with following:

> nmha@pine64:~$ urh
Will not regenerate UI, because script cant be found. This is okay in release.
Could not find C++ extensions, trying to build them.
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/urh/main.py", line 61, in main
    import urh.cythonext.signalFunctions
  File "__init__.pxd", line 155, in init src.urh.cythonext.signalFunctions (src/urh/cythonext/signalFunctions.cpp:21885)
  File "/usr/lib/python2.7/dist-packages/numpy/__init__.py", line 180, in <module>
    from . import add_newdocs
  File "/usr/lib/python2.7/dist-packages/numpy/add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "/usr/lib/python2.7/dist-packages/numpy/lib/__init__.py", line 8, in <module>
    from .type_check import *
  File "/usr/lib/python2.7/dist-packages/numpy/lib/type_check.py", line 11, in <module>
    import numpy.core.numeric as _nx
  File "/usr/lib/python2.7/dist-packages/numpy/core/__init__.py", line 14, in <module>
    from . import multiarray
ImportError: cannot import name 'multiarray'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/urh", line 11, in <module>
    load_entry_point('urh==1.6.1.9', 'console_scripts', 'urh')()
  File "/usr/local/lib/python3.5/dist-packages/urh/main.py", line 67, in main
    os.chdir(os.path.join(src_dir, "urh", "cythonext"))
FileNotFoundError: [Errno 2] No such file or directory: '/usr/src/urh/cythonext'

Please kindly help, is it possible to run URH under this Armbian? 
Thank you very much in advance!

jopohl, 2017-03-31:
Running URH on ARM sounds like fun! :wink: 

From the log you posted it seems, that numpy is only present for python 2.7 so URH can not compile its extensions properly.

What does ``` python3 -c "import numpy" ``` give you?

ngmacha, 2017-03-31:
Oh, got your reply while already closing down everything in the office - I'll check and reply once get home. 
Have a nice weekend!
Mac Ha

ngmacha, 2017-03-31:
Here is the output of `python3 -c "import numpy"` command:

> nmha@pine64:~$ python3 -c "import numpy"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/numpy/__init__.py", line 180, in <module>
    from . import add_newdocs
  File "/usr/lib/python2.7/dist-packages/numpy/add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "/usr/lib/python2.7/dist-packages/numpy/lib/__init__.py", line 8, in <module>
    from .type_check import *
  File "/usr/lib/python2.7/dist-packages/numpy/lib/type_check.py", line 11, in <module>
    import numpy.core.numeric as _nx
  File "/usr/lib/python2.7/dist-packages/numpy/core/__init__.py", line 14, in <module>
    from . import multiarray
ImportError: cannot import name 'multiarray'

However:

> nmha@pine64:~$ sudo apt-get install python3-numpy
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-numpy is already the newest version (1:1.11.0-1ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

Or there is another numpy package to be installed? Oftentimes all these Linux packages are very confusing, especially to mostly-Windows folks like me!

ngmacha, 2017-03-31:
Having a second look at my post above, it seems strange to me that although I explicitly invoked 'python3', it still looks into "/usr/lib/python2.7/...".

jopohl, 2017-03-31:
This is also seemed strange to me in your first post. Seems like your numpy installation somehow was performed for python 2 instead python 3. Could the Ubuntu package be broken? Can you try installing numpy with:
```
pip3 install numpy
```
and see if you can run ``` python3 -c "import numpy" ``` afterwards?

ngmacha, 2017-03-31:
I think initially numpy was installed via `apt-get install python3-numpy` following instruction in the README. 
Here is the output of `pip3 install numpy` (BTW, the "running setup.py install for numpy" took very long time, maybe 30 minutes - is that normal?...):

> nmha@pine64:$ pip3 install numpy
Collecting numpy
  Using cached numpy-1.12.1.zip
Building wheels for collected packages: numpy
  Running setup.py bdist_wheel for numpy ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-woqeoam6/numpy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" bdist_wheel -d /tmp/tmph6a654g5pip-wheel- --python-tag cp35:
  Running from numpy source directory.
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  error: invalid command 'bdist_wheel'
  Failed building wheel for numpy
  Running setup.py clean for numpy
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-woqeoam6/numpy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" clean --all:
  Running from numpy source directory.
  `setup.py clean` is not supported, use one of the following instead:
    - `git clean -xdf` (cleans all files)
    - `git clean -Xdf` (cleans all versioned files, doesn't touch
                        files that aren't checked into the git repo)
  Add `--force` to your command to use it anyway if you must (unsupported).
  Failed cleaning build dir for numpy
Failed to build numpy
Installing collected packages: numpy
  Running setup.py install for numpy ... done
Successfully installed numpy-1.11.0
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

Then:

> nmha@pine64:$ python3 -c "import numpy"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/numpy/__init__.py", line 180, in <module>
    from . import add_newdocs
  File "/usr/lib/python2.7/dist-packages/numpy/add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "/usr/lib/python2.7/dist-packages/numpy/lib/__init__.py", line 8, in <module>
    from .type_check import *
  File "/usr/lib/python2.7/dist-packages/numpy/lib/type_check.py", line 11, in <module>
    import numpy.core.numeric as _nx
  File "/usr/lib/python2.7/dist-packages/numpy/core/__init__.py", line 14, in <module>
    from . import multiarray
ImportError: cannot import name 'multiarray'

So, still the same errors... Weird.

BTW, I have gqrx installed, which runs fine. So what might be problem here?..

ngmacha, 2017-03-31:
Looks like the numpy for python3 is installed, indeed:

> nmha@pine64:$ ls /usr/lib/python3/dist-packages/numpy
add_newdocs.py  distutils         __init__.py  matrixlib    testing
compat          dual.py           lib          polynomial   tests
__config__.py   f2py              linalg       __pycache__  version.py
core            fft               ma           random
ctypeslib.py    _import_tools.py  matlib.py    setup.py

Very strange... So why python3 still looks into the 2.7 library?!

jopohl, 2017-03-31:
Numpy needs to be compiled from source if you install it via setup.py. That is why the Readme recommends to install it via package manager (``` python3-numpy ``` ) or pip3 before installing URH.

It is indeed very strange that Python3 looks into Python2.7 directory for numpy. Do you need numpy for Python2? Otherwise you may try ``` apt-get uninstall python-numpy ``` or ``` pip uninstall numpy ```, depending on the way how you installed numpy for python2.

ngmacha, 2017-04-01:
I didn't install numpy for python2 myself, and not quite sure when was it been installed. But you are right - I removed it (BTW, the correct command was `apt-get remove python-numpy` ) and URH started!! So I'll have to check later which program stopped to work without the pyhton2-numpy, and how to have both version 2 and 3 to coexist. If I find out and you'll find useful, I could share the how-to for you to include into the README.

Though URH is starting now, I have several other problems - should I open new issue for each of them? Let me just list two of them here first - maybe they are all known issues for you (I'm a bit new to Github, so please advise if I shouldn't do this in the future - do you have then a discussion group where these can be raised):
1. The LCD display on my Pine64 board is having only resolution of 1024x600 while URH and most of its dialogs opens by default with the height large than 600 pixels. There are no vertical scroll bars, making the buttons and options at the bottom inaccessible - so every time opening a new dialog I have to resize them. Is it possible to enable vertical scrolls, or set the default screen size to which the windows would fit to?
2. Somehow the Devices has only RTL-TCP enabled, and all other devices options are greyed out. I enabled it uder `~/.config/urh.ini`, but starting reset everything back to disabled. I'm sure that my RTL-SDR is correctly installed, as gqrx is running fine. But do I need gnuradion to be able to use it under URH, as I  was unable to install the full gnuradio yet?
Thank you.

jopohl, 2017-04-01:
I just wrote a fix 6c0267651bdf062807683de7f67e14b7dea56fe2 for low resolutions. URH should be way better accessible on your screen now! You can check out master to have access to this fix or just wait until I make the next release.

For devices you should ensure that you have the required libs installed on your system. What does ``` locate rtlsdr.so ``` say? You should not enable devices manually in ini. Native backend should work for you if you installed rtlsdr.so, you can use the rebuild button in device options for this.

If you have further issues do not hesitate to open a new issue for each of them. This way things stay nice and clear: Each issue should focus on one problem.

closed 2017-04-03 by ngmacha