jopohl/urh #221: cannot enable HackRF

compr00t, 2017-03-31:
I'm trying to use URH on OSX (macOS Sierra) together with my HackRF but I cannot enable the needed device in the settings although I have installed the needed library:

```
$ file /opt/local/lib/libhackrf.dylib
/opt/local/lib/libhackrf.dylib: Mach-O 64-bit dynamically linked shared library x86_64
```

And I can use the hackrf through this library without any problem:

```
$ hackrf_info
Found HackRF board 0:
USB descriptor string: 0000000000000000909864c82c4534cf
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x00664f5e
Serial Number: 0x00000000 0x00000000 0x909864c8 0x2c4534cf
```

The setup even works with gnuradio-companion. But URH is still telling me it is disabled. I even tried rebuilding with the added path of `/opt/local/lib`but without success. What am I missing?

jopohl, 2017-03-31:
Can you post the command line output, when to rebuild the device extensions in the options?

compr00t, 2017-03-31:
sure.

```
$ urh
Will not regenerate UI, because script cant be found. This is okay in release.
/var/folders/q3/hzwjfdw92lbg768nnj4n_h1w0000gn/T/rtlsdr_set_tuner_bandwidthn_mwbrc5.c:1:1: warning: type specifier missing, defaults to
      'int' [-Wimplicit-int]
main (int argc, char **argv) {
^
/var/folders/q3/hzwjfdw92lbg768nnj4n_h1w0000gn/T/rtlsdr_set_tuner_bandwidthn_mwbrc5.c:2:5: warning: implicit declaration of function
      'rtlsdr_set_tuner_bandwidth' is invalid in C99 [-Wimplicit-function-declaration]
    rtlsdr_set_tuner_bandwidth();
    ^
2 warnings generated.
ld: warning: directory not found for option '-L,'
ld: library not found for -lrtlsdr
clang: error: linker command failed with exit code 1 (use -v to see invocation)
Trying fallback for rtlsdr
/var/folders/q3/hzwjfdw92lbg768nnj4n_h1w0000gn/T/rtlsdr_get_device_namewtexdn4q.c:1:1: warning: type specifier missing, defaults to
      'int' [-Wimplicit-int]
main (int argc, char **argv) {
^
/var/folders/q3/hzwjfdw92lbg768nnj4n_h1w0000gn/T/rtlsdr_get_device_namewtexdn4q.c:2:5: warning: implicit declaration of function
      'rtlsdr_get_device_name' is invalid in C99 [-Wimplicit-function-declaration]
    rtlsdr_get_device_name();
    ^
2 warnings generated.
ld: warning: directory not found for option '-L,'
ld: library not found for -lrtlsdr
clang: error: linker command failed with exit code 1 (use -v to see invocation)
/var/folders/q3/hzwjfdw92lbg768nnj4n_h1w0000gn/T/hackrf_initj0717jei.c:1:1: warning: type specifier missing, defaults to 'int'
      [-Wimplicit-int]
main (int argc, char **argv) {
^
/var/folders/q3/hzwjfdw92lbg768nnj4n_h1w0000gn/T/hackrf_initj0717jei.c:2:5: warning: implicit declaration of function 'hackrf_init' is
      invalid in C99 [-Wimplicit-function-declaration]
    hackrf_init();
    ^
2 warnings generated.
ld: warning: directory not found for option '-L,'
ld: library not found for -lhackrf
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Skipping native support for hackrf
```

jopohl, 2017-03-31:
Looks like clang does not search for the dylib in the right directory. Can you try:
1. Go to Options -> Device -> Enter " /opt/local/lib" under library directories and press rebuild. You may need to start URH as root for this, depending on where you installed it.
2. If this does not work: cp /opt/local/lib/libhackrf.dylib /usr/local/lib/libhackrf.dylib and build again

compr00t, 2017-03-31:
Already tried option 1, without success. But option 2 did the trick, thanks for your help!

closed 2017-03-31 by compr00t

jopohl, 2017-03-31:
You're welcome @compr00t . I just wrote a small fix in 1d0e5cc9ac67cdd72753da084be1d2ffbc2f5d28, so option 1 will work in the future.